micronucleus/micronucleus #61: AUTO_EXIT_MS at 250ms silenced my Digispark :)

jej, 2015-04-15:
Hi,
I wanted to speed-up the boot time and I played with the AUTO_EXIT_MS delay, so that the Digispark waits less before starting the program.
At 1000ms it was ok. I tried 250ms and now it is not responding to uploads........ I tried to tweak down some delay in the commandline (CONNECT_WAIT) but nothing helped.
Any idea how I could recover it? without guru hacks (fuses, isp and so on...)
In another repository (https://github.com/andihofmeister/micronucleus-t85/blob/master/firmware/bootloader-timer-mode.h) I saw that there is a way to warn against that mistake:
```
    #if AUTO_EXIT_MS < (MICRONUCLEUS_WRITE_SLEEP * (BOOTLOADER_ADDRESS / SPM_PAGESIZE))
    #    warning "AUTO_EXIT_MS is shorter than the time it takes to perform erase function - might affect reliability?"
    #    warning "Try increasing AUTO_EXIT_MS if you have stability problems"
    #endif
```
Maybe that's not up-to-date with the current source of micronucleus, but in that condition AUTO_EXIT_MS should be >= 792.

Thanks for your help.

cpldcpu, 2015-04-16:
The timer is halted during the erase, so the condition above is not really relevant.

The problem with your configuration is probably that the device times out before USB-enumeration is completed. The enumeration time seems to depend on operating system and USB topology. You could try to connect the device to a port without a hub. USB3.0 also seems to be slower.

There are a zillion ways to break the bootloader. It is impossible to guard against all configuration mistakes. The safest way is to test it without disabling reset so the bootloader can be tested without briking the device.

jej, 2015-04-16:
Thanks for you answer.
I will try a bit more with "USB enumeration" so...
> timer is halted

As it is just a compile time calculation, is the timer involved in this estimation?
> There are a zillion ways to break the bootloader

Yes sure!!! Maybe a little warning in the comments could prevent some errors for novices like me. I will propose a patch.
I didn't know it was possible to change the reset configuration. I will look at this. Thanks.

closed 2015-05-30 by cpldcpu