micronucleus/micronucleus #63: Documentation says 6380 bytes in 1.11, but uploader only says 6330

hachi, 2015-04-17:
> Please plug in the device ...
> Press CTRL+C to terminate the program.
> Device is found!
connecting: 16% complete
connecting: 19% complete
connecting: 22% complete
connecting: 25% complete
connecting: 28% complete
connecting: 33% complete
> Device has firmware version 1.11
> Available space for user applications: 6330 bytes
> Suggested sleep time between sending pages: 8ms
> Whole page count: 99  page size: 64
> Erase function sleep duration: 792ms
parsing: 33% complete
parsing: 50% complete
> Program file is 16 bytes too big for the bootloader!

cpldcpu, 2015-04-17:
The size of the bootloader can very depending on the compile-time options that were chosen and the version of the compiler that was being used. Did you compile the bootloader by yourself?

hachi, 2015-04-17:
I didn't. I'm using micronucleus-1.11-ledpb1-upgrade.hex

I'll give a go to the raw one.

cpldcpu, 2015-04-17:
Yes, the LED-code needs some extra-bytes that will add another memory page.

You can also try MN v2.0, which is much smaller -> testing branch

hachi, 2015-04-17:
I just tested again with micronucleus-1.11-upgrade.hex and still only get 6330. Is there a file in the upgrade/releases/ directory that gives me 6380?

I'll also give a shot to 2.0.

hachi, 2015-04-17:
Well, I may need some time to try out 2.0, I don't have a toolchain to build it with yet.

closed 2015-05-30 by cpldcpu