micronucleus/micronucleus #64: Attiny167

Collie147, 2015-06-13:
Tried writing 2.0b hex file to attiny167. Worked successfully but chip doesn't respond at all now.  Haven't tried high voltage reset just yet but I'm wondering could it be the fuse settings? Or perhaps that the d+ pin on the bootloaderconfig.h file is listed as portb4 which is also xtal1???

cpldcpu, 2015-06-15:
You are right, the configuration was indeed off. I changed it to follow the Digispark Pro design. It should work now. Can you try again? You will need an external 16MHz xtal, though.

cpldcpu, 2015-06-15:
Btw, currently the makefile does not contain any fuse setting for the ATtiny167. You need to change it manually.

Collie147, 2015-06-15:
Thanks Tim

Unfortunately I can't write to the chips at all.  Tried it on both of
them.  I'll research into HVSP on the 167 and see if I can get it to work.
I'll let you know the updated build works ok.

Thanks again

cpldcpu, 2015-06-15:
Well, since the micronucleus makefile does not write to the fuses of the ATtiny167, I have a hard time imagining how the firmware would be able to to render the device unresponsive? Let me know if HVSP helps.

closed 2015-06-20 by cpldcpu