micronucleus/micronucleus #65: Wiki Topic: Flashing both micronucleus and user program in one "hex"

Zitt, 2015-07-09:
Could someone in the know "document" how a user would go about programming the micronucleus bootloader and the user program with one hex file?

Example Use Case:
End-Product is a hobby device with  a built-in user program to control an LED.
During manufacturing of end-product; would like to supply the PCB house with a single .hex file which contains both the bootloader and the user program so they only have to "socket"/program once.

deladriere, 2015-07-09:
:+1: 
I also want to do that !
 (not only with micronucleus but also with Optiboot on Arduino)

Bluebie, 2015-07-09:
The safest and most commonly used strategy is this:

1. setup micronucleus on your destination chip, but do not disable ISP reset pin.
2. use USB to upload default userspace program via micronucleus as usual.
3. use ISP programmer to dump firmware .hex.
4. flash resulting combined hex to end user devices, disabling reset pin if you need to.

That's how DigiSpark default firmware was derived.

closed 2015-07-09 by Bluebie

Zitt, 2015-07-10:
added a WikiPage based upon your info:
https://github.com/micronucleus/micronucleus/wiki/Flashing-both-micronucleus-and-a-user-program-with-one-hex

fraser125, 2015-07-10:
There is a much more standard way. I'm not a Makefile GURU but here is a link to one that merges a bootloader with the application into a single hex file. https://github.com/evo-drone/drone-1.0/blob/master/v2.1/merge/Makefile 

It's a command line program called srec_cat Take a look at the MAN page for it http://srecord.sourceforge.net/man/man1/srec_cat.html 

I hope this helps.

Bluebie, 2015-07-10:
TLDR: nope don’t do that.

----

That technique is incompatible with micronucleus and every other bootloader for the attiny85 and other avr chips which have no hardware bootloader support.

There is no simple way to merge a firmware and bootloader on the host side for these chips. The bootloader hacks the interrupt vector table of the user firmware to jump through itself, and the position of the bootloader is different in different versions of micronucleus. Any code you wrote that would generate the correctly hacked file would be tightly coupled to just one build of micronucleus on just one platform in just one configuration.

Just do it how I said.

sowbug, 2015-12-10:
Long version of @Bluebie's TLDR.

Concatenating the two hex files into a single hex file isn't enough. Micronucleus needs to run every time the device starts, even if you have it configured to run only on reset or watchdog timeout. So if you flash just app+bootloader, the bootloader never has a chance to run. You'll have a working app, but it will lose bootloader functionality.

At the following block of code in the `micronucleus` command-line tool, you can see it patches the app it's uploading to jump to the bootloader: https://github.com/micronucleus/micronucleus/blob/master/commandline/library/micronucleus_lib.c#L198 It writes the original 2 or 4 bytes of the program (a jump command) to the very very very end of the space available for the app so that it can remember how the app starts: https://github.com/micronucleus/micronucleus/blob/master/commandline/library/micronucleus_lib.c#L238

Incidentally, this is where the bootloader uses that jump command to start the app: https://github.com/micronucleus/micronucleus/blob/master/firmware/main.c#L225

There's another problem that the bootloader prefers to live way up top in program memory, rather than at the end of the memory used by the app, but you get the point.

The reason the two-step upload-download is the most practical solution way to go is because (1) it's actually simply considering the relatively rare number of times you need to create the factory image, and (2) micronucleus might change its approach to app-patching and resuming, in which case whatever tool you write that duplicates the patching will break, and you'll have to reverse-engineer it all over again.

I suppose the extra style points version of the fix to this bug would be a patch to the `micronucleus` command-line tool that adds the option to spit out a hex file locally rather than uploading the hex bytes. You'd have to keep a copy of bootloader.hex around, or else have a device handy to read the correct bootloader. That's a lot of work, though, to replace a single avrdude command.

sowbug, 2015-12-10:
For a real-world example, see https://github.com/sowbug/weblight/blob/master/firmware/Makefile, explained in https://github.com/sowbug/weblight/blob/master/README.md.