micronucleus/micronucleus #69: Idea: updating micronucleus by itself

cehteh, 2015-09-01:
disclaimer: Dunno if this will work, I may give it a try, comments/opinions are welcome.

The Plan:
Write a program 'updater' which wraps a new version of micronucleus and flash that to the chip.
Upon the next boot this program will self-program the new micronucleus to the begin of the flash.

the tricky parts are that the updater (including the new micronucleus image) should be aligned to the end of the flash or must at least start after the to-be end of the new micronucleus after self-flashing.
Also upon successful flashing it needs to disable itself to prevent reboot/reflash loops.

Bluebie, 2015-09-01:
I can't tell if you're asking us to change updater to work differently or if you just don't know that we already implemented this idea years ago?

Please clarify.

cpldcpu, 2015-09-01:
I believe he just doesn't know. The update tool has not been migrated to
2.x yet, that's why it isn't in the current repository.

Bluebie, 2015-09-01:
Ooh right I forgot that never got sorted. Yeah we should get an upgrader tool! heh

gdamjan, 2016-01-23:
So, is upgrading micronucleus even possible now?

w-A-L-L-e, 2016-01-23:
You can get an older version of upgrade tool in v 1.11 branch...

hafkensite, 2016-11-01:
I see a upgrade make target in the `/firmware` directory, but do not understand what it does. I expected it to output a new .hex file that I can call with the micronucleus commandline app. 
It flashing the 2.x version over a digispark with 1.x possible?

dcbishop, 2016-11-15:
@hafkensite:

The upgrade target in /firmware isn't what you want. You need to get the 1.11 branch and use the ruby script in upgrade/generate-data.rb and point it at the bootloader generated by the newer 2.xx branch.

"ruby generate-data.rb ~/src/micronucleus-2.03/firmware/main.hex"

This should produce the upgrade.hex file you use the micronucleus cli to run.

You can also look at changing the BOOTLOADER_ADDRESS as mentioned in the upgrade/readme.txt to account for the size but it worked fine for me without that.

krzcho, 2017-01-19:
I did it as described using master's t85_default.hex as base, got upgrade.hex, uploaded using cli but then after reboot device still reports 1.10 firmware :( Any clues?

carwe, 2017-06-07:
Is there a reason to not just add the generate-data.rb from v1.11 script to the actual v2 version? At least for me it worked just fine so far with Ruby 2.1, generating a upgrader-payload containing the v2.03 bootloader, which I flashed then via the v2.03 commandline tool to a attiny85 with "v1.60" (was installed on a digithumb I bought in China) bootloader.

cpldcpu, 2017-08-20:
The reason is basically that nobody tested compatibility of the ruby script and the newer versions of MN. I planned to implement a different build flow for upgrade that does not rely on ruby (by using AVR link tools), but that newer materializes.

So, I am merging @Ho-Ro pull request in the hope it works for everyone.

closed 2017-08-20 by cpldcpu