micronucleus/micronucleus #71: micronucleus on attiny45

OzanBarsal, 2015-10-09:
Hello i'm not able to get the micronucleus bootloader on attiny45, i'm pretty much a beginner, i tried to flash attiny85 build directly which may seem stupid,  and it said the address of the beginning line on the hex file is out of the memory space. It does make sense, i checked the hex file and the addresses wasn't starting from 0, i guess the firmware places itself starting from an address which is already beyond the attiny45's memory size.

I'm not able to build the source code of the micronucleus for attiny because i'm a beginner, are there any ready to use builds or can anyone build me one?

My programmer works fine and i just need a bootloader to get my attiny 45 working with usb.

Thanks.

Bluebie, 2015-10-09:
I don't currently have build tools installed. Perhaps Tim could do a custom build. I know there was an ancient tiny45 build at some point which you may be able to find on the digistump/digispark forums. It should still be compatible with digistump arduino tools.

The reasoning on why we don't have builds for tiny45 is because in most instances, tiny85 is cheaper to purchase than tiny45, and better or equivalent in every way. Economies of scale just don't work out in favour of tiny45. I'd suggest if you are at all able, replace the chip with a tiny85. You'll have a much better time. 6kb is already quite small for many projects.

cpldcpu, 2015-10-09:
There you go:

https://github.com/micronucleus/micronucleus/blob/master/firmware/releases/t45_default.hex

It's not terribly useful, though, as you will have barely more than 2kb of flash left. As Jenna suggested, it would be better to get an ATtiny85.

closed 2015-10-09 by cpldcpu