micronucleus/micronucleus #72: Micronucleus and other USB functionality

GGreenwood, 2015-10-21:
Hi all,

I have a project that I would like to add bootloader functionality to. However, this project already has its own USB descriptor and functionality as a gamepad. I know that V-USB supports switching between different descriptors at runtime, but I'm not sure how that interacts with micronucleus.

I'd like to be able to do the following:
* On boot, the user can hold a button to enter bootloader mode. Otherwise it would act normally.
* I would like to be able to update the USB descriptor for my gamepad functionality along with the rest of the program.
* It would be nice(but not necessary) if I could reuse the V-USB library existing in the micronucleus bootloader without needing to reimport it in the user program.

How compatible are these goals with micronucleus? I'd really like to use this project(the post on interrupt-free V-USB inspired me to add USB functionality), but I'm not familiar enough with the project to know whether they're possible.

cpldcpu, 2015-11-03:
Hi,

your functioanlity can be easily implemented by configuring micronucleus only to be activated when a button is pressed. Check out the configuration files, it is described there. Let me know if there are addiutional quest

On reusing the same V-USB for microncleus and the main program: This seems to be a fairly obvious improvement, but unfortunately it is extremely difficult to realize in practice. 

To name two reasons
 * V-USB uses a lot of compile time options, so small changes in your code could yield a completely different binary of V-USB. It would be very difficult to find a matching superset of options to serve both your code in micronucleus. Also, the MN implementation of V-USB has been heavily modified.

* Using the same code for the bootloader and user program would mean that the bootloader would have to overwrite itself. This is not easy...

GGreenwood, 2015-11-10:
Thank you for the answer, it clarified a lot.

The joystick that I'm using implements a proprietary communication protocol and I have my own library designed to communicate with it. Therefore, I believe that I would need to modify the bootloader source to include a custom ENTRYMODE that would reference my own code. This wouldn't be too hard, though.

I see what you mean about the compile time options. I'll just duplicate a little bit of code, it wouldn't be worth the effort. 

I'll go ahead and close this issue, thanks for the advice!

closed 2015-11-10 by GGreenwood