micronucleus/micronucleus #73: nanite841 or nanite85 not working

nevvkid, 2016-01-25:
hi. 
i bought a couple of nanite boards at watterott recently and tried to get them working on os x (as well tried them on linux, windows 10) and none of the get recognised from the usb device manager. 
are there known issues with the bootloader / boards that they are not working properly? do i have to flash a new bootloader with an isp?

cpldcpu, 2016-01-25:
It seems a bit odd that all combinations are not working. Well, no, there is no known issue that could explain this. Did you hit the reset button on the Nanite?

Micronucleus sometimes works better if it is connected directly to a primary port instead of a hub.

nevvkid, 2016-01-26:
yeah. for me as well. i tried on three different computers and always without hub, different cables. maybe you find the time write a little how to guide in 5 sentences for the nanite boards?

## OSX
1. Download in install latest arduino [IDE from arduino.cc](https://www.arduino.cc/en/Main/Software)
2. Add Wattuino board support package in "Preferences" [get wattuino board support package url here](https://github.com/watterott/wattuino/tree/master/software/Arduino#watterott-board-support-package)
3. Download and install Watterott AVR Boards in "Tools/Board/Board Manager"
4. Restart IDE
5. Write example sketch
6. compile & upload 
< this is where it fails > i press reset when uploading is shown > should i hold the reset or press is only once? i always get a timeout. in the OSX USB device manager the boards never show up. same thing happens on windows or linux on different machines... 

the nanite841 led flashes fast if the board is connected to the computer. while pressing reset it turns off.

are there any tipps to debug? upload code in a different way from sublime / terminal / isp? get the usb working? drivers i need to install on OSX or patch some files? Running OS X 10.10.5 currently.

nevvkid, 2016-02-03:
Found the solution. We had to recompile the micronucleus files included in the wattuino board manager files. They currently don't work with 64bit os x.

closed 2016-02-03 by nevvkid