micronucleus/micronucleus #76: Question about eeprom

simonvu14, 2016-05-26:
Hi, 
I am kinda noob and this is my silly questions:
- Is it possible to "burn" eeprom of Tiny85 along with flash using micronucleus?
- Is it possible to "burn" eeprom of Tiny85 without touching the flash area, using micronucleus?
Thanks,
Simon Vu

cpldcpu, 2016-05-26:
Micronucleus currently does not have EEPROM writing functionality, but with some effort it should be possible to add it.

simonvu14, 2016-06-03:
Thanks for fast reply.

I have another question: does uploading new hex file using Micronucleus erase the existing content of EEPROM? I am thinking of using another sketch, such as digiUSB, to write/modify EEPROM. And when it's done, I could upload the original sketch back.

Anyway, if eeprom writing function could be implemented, how can I do it?

I am making a custom keyboard using tiny85, and a software on PC to change the function of buttons. And EEPROM is the best way to modify and store settings.

Thanks again.

Bluebie, 2016-06-04:
Yes this will work fine. Micronucleus does not modify EEPROM. You might like to look in to combining the functions of EEPROM writing and keyboard in to one firmware. It's certainly possible to send messages to keyboards through the HID system to send configuration information like that.

simonvu14, 2016-06-04:
Yes, it would, theoretically. The thing is my program nearly comes to the limit in size about 5800 Bytes, no more space for anything else. Well, I could use mega328 or mega32u4 which has more features, but what is the fun in it.

Bluebie, 2016-06-04:
Fair enough! that seems like a reasonable strategy in that case!

phw198, 2016-07-26:
Hi! Would it be possible to reopen this issue/request? I could really do with being able to flash the EEPROM separately from the firmware, to store data like serial numbers that are unique to a digispark, but aren't changed through the firmware.

cpldcpu, 2016-07-27:
Well, I think it is quite a bit of work to add some kind of generic solution for that, but if you have something very specific in your mind, it could be added more easily.

Just add another command:
https://github.com/micronucleus/micronucleus/blob/master/firmware/main.c#L77
https://github.com/micronucleus/micronucleus/blob/master/firmware/main.c#L166

phw198, 2016-07-28:
Thanks - I'm pretty new at all this, but I've had a look around the code. 

I think I'd basically want to add a "--eeprom" argument to the command line tool's main function. Then have a conditional on https://github.com/micronucleus/micronucleus/blob/master/commandline/micronucleus.c#L271 and call a new function micronucleus_writeEeprom(). This would simply pass the eeprom hex buffer to a new command in the firmware that could just loop round the bytes in the buffer and use eeprom.write() from eeprom.h to flash it.

If this sounds sane, I'm happy to give it a go and make a pull request if it works...

cpldcpu, 2016-07-31:
Yes, that's basically how it goes. The tricky part is breaking down your data into chucks of 4 bytes that can be trasnferred via USB. I would recommend to write only one byte at a time. The EEPROM writes are nonblocking so that can be implemented straight forward in the command loop.

plasticassius, 2016-08-03:
I have a few thoughts to add to this. Firstly, you don't need anything from micronucleus to read and write the EEPROM. The functions in avr/eeprom.h let you read and write from any loaded firmware. However, if you update micronucleus, you will erase the EEPROM contents due to the default setting of the EESAVE fuse (preserve EEPROM on Chip Erase). To preserve it, you must program the EESAVE fuse.

Also, you can achieve something similar to what the EEPROM does by placing configuration information into the micronucleus hex file itself. I've added an option to put a 16 bit serial number into the micronucleus image in pull request #80. I return that serial number from the loaded firmware as part of it's USB initialization so that each device can identify itself when a connection is established.

sowbug, 2016-09-01:
@plasticassius: if I'm understanding @phw198's request, the solution you have in mind won't handle this case:

- Want to manufacture 1,000+ of something.
- Have micronucleus bootloader and your own program combined in a single .hex that you send to DigiKey or Mouser or your PCBA house for programming. They do the fuse/flash for you. For the price you negotiated, they won't modify the hex or do unique-per-board EEPROM programming; it's just an opaque blob to them.
- You get back your assembled boards.
- Before shipping the boards off to customers you stamp a serial number into the EEPROM by briefly shoving them into a USB port.

It'd be really nice to be able to do this via micronucleus, rather than either devoting some of your own program space to a custom serial-number-writing command handler, or dragging out an ISP and firing up avrdude. It could be a Makefile option if someone didn't want to spend the bytes in the bootloader.

sowbug, 2016-09-01:
@phw198 please let me know if you're still planning on implementing this patch.

phw198, 2016-09-01:
@sowbug Yes, you've summarised my requirement pretty well and my apologies for not posting an update sooner. I did have a stab at patching it, but I rather fear I'm a bit out my depth :) The eeprom.write() function I had seen appears to be in a C# header which isn't compatible. I did find a C version of eeprom.h, but this is beyond my abilities plus it required quite a number of additional header files to be referenced, which will no doubt bloat the firmware beyond its currently allocated space.

I was going to settle for my own command line tool to write to the EEPROM so, as you say, I can pop in the device, pass the serial number as an argument on the CLI and disconnect. Would be nice to have this built in to the micronucleus CLI and Makefile, but understand if it's too much work...

zeevro, 2017-06-05:
@simonvu14 You mentioned a custom keyboard based on the attiny85. Did you finish that project in the end? Can you publish the code?

closed 2017-07-30 by cpldcpu