micronucleus/micronucleus #77: Trouble uploading on windows 10

Cacowned, 2016-06-16:
Hi,

I have 2 digisparks with micronucleus v1.11 on them. When I try to use the micronucleus commandline tool to update either one of them, they are not found and nothing happens, the device goes to the user program after 5 seconds (preloaded with blink sketch). In those 5 seconds, the device is found by windows (I installed the drivers with zadig and/or digistump installer), but somehow the commandline tool can't see the device.

I took one of the devices to a windows 7 machine, installed the driver with zadig, downloaded the micronucleus repo, created a toolchain to build a v2 bootloader, built the v2 bootloader and flashed it to the device using the micronucleus commandline tool. So, the device works at least, yay!

Back to the windows 10 machine. No user program, so the device stays in micronucleus mode and windows sees the device (Device manager shows it, Zadig shows it), but when I start, for instance, micronucleus --erase-only and then plug-in the device, nothing happens.

What am I missing here? I'm sure the device is not bricked, it still works on the win7 machine...

cpldcpu, 2016-06-20:
Not quite sure what is going on here. There was indeed an issue with Win10 that did prevent proper usb enumeration for micronucleus, but this is fixed in the latest version.

Did you try the other way around? First plugging in the device, the starting the commandline tool? A freshly installed MN does not time out.

Cacowned, 2016-06-20:
Oh, the commandline tool doesn't time out, neither does the device with only micronucleus installed. But a device with a userprogram loaded automatically starts the program after 5 seconds when the device doesn't get a signal from the commandline tool.

But to answer your question, yes, I tried that as well, and to no avail.

mjbcopland, 2016-06-21:
Also experiencing the same issue. Micronucleus shows in device manager but the commandline tool does not detect it. It appears as though [*usb_get_busses()*](https://github.com/micronucleus/micronucleus/blob/master/commandline/library/micronucleus_lib.c#L42) returns a single bus, but [*bus->devices*](https://github.com/micronucleus/micronucleus/blob/master/commandline/library/micronucleus_lib.c#L47) is *NULL*.

Adding *printf* statements like so:
```
...
struct usb_bus *bus;
fprintf(stderr, "1");
for (bus = busses; bus; bus = bus->next) {
  struct usb_device *dev;
  fprintf(stderr, "2");
  for (dev = bus->devices; dev; dev = dev->next) {
    fprintf(stderr, "3");
    /* Check if this device is a micronucleus */
...
```
produces the output "121212121212..."

mjbcopland, 2016-06-21:
I've modified the helper library to use libusb 1.0.20 from [libusb.info](http://libusb.info/). It potentially still needs work (update the Windows driver accordingly?) but it now recognises the USB device properly. See [here](https://github.com/mjbcopland/micronucleus/tree/libusb-1.0.20) for the new version.

Cacowned, 2016-06-21:
NIce! Will try it out when I get home tonight.

Cacowned, 2016-06-21:
Your fix works for me as well, thank you for looking into this.

cpldcpu, 2016-06-21:
@mjbcopland 

Thanks a lot! Looks like I have to change the compiling environment to MSYS2 for the latest libusb. Can you submit a pull request for your changes?

mjbcopland, 2016-06-22:
Sure. I've added another commit which now iterates through the USB devices manually, removing the function which is "not intended for use in real applications". I haven't changed the driver installer so the new version needs *libusb-1.0.dll* in the same folder as *micronucleus.exe*.

PR #83 (2017-08-20): commandline: compiled using libusb compatibility layer
Using https://github.com/libusb/libusb-compat-0.1 with libusb 1.0.20

Works for me, should fix #77 and #81 for others.
closed 2016-07-25 by Cacowned