micronucleus/micronucleus #81: Not compiling in fedora 23 x64

rm3nchaca, 2016-07-29:
It gives a lot of errors because of usb.h:

Executing make in Fedora 23 x64 gives a lot of errors like this:

In file included from library/micronucleus_lib.c:30:0:
library/micronucleus_lib.h:68:3: error: unknown type name ‘libusb_device_handle’
   libusb_device_handle *device;
.
.
.
.

I 've compiled it with a little workaround in micronucleus_lib.h:
```
#if defined WIN
  #include <libusb.h>     // this is libusb, see http://libusb.sourceforge.net/
#else
  //include <usb.h>        // this is libusb, see http://libusb.sourceforge.net/
  #include <libusb-1.0/libusb.h>
```
And also in the Makefile:

```
#CFLAGS  = $(USBFLAGS) $(LIBS) -I$(INCLUDE) -O -g $(OSFLAG)
CFLAGS  = -lusb-1.0 $(LIBS) -I$(INCLUDE) -O -g $(OSFLAG)
```

I have no experience with make neither compiling, maybe just adding an additional condition depending of the linux distro solve the problem.

In the meantime, you can use this if you are getting trouble compiling in fedora.

Greetings!

cpldcpu, 2016-07-31:
This seems to be a version conflict of libusb, not specifically limited to fedora.

rm3nchaca, 2016-07-31:
You're right, the definitions in the .h file are different and in Fedora
you need to specify de version of libusb.

Schm1tz1, 2016-10-21:
Does this problem still persist ? I have just been compiling the recent version on FC23 x64 without any problems. Maybe a version conflict ?

cpldcpu, 2016-10-22:
As mentioned, this rather seems to be related to the libusb version that's installed. I'll close this issue to avoid confusion.

closed 2016-10-22 by cpldcpu