micronucleus/micronucleus #86: bootloader hang when using external power (no usb data) on RESET

Palatis, 2016-10-25:
I recently build a small circuit that use the USB to charge a supercapacitor, and run from that.
The circuit basically looks like this:
[image]

I had `#define AUTO_EXIT_MS 1000` in bootloaderconfig.h, the rest are the same.

The problem is, when I first plug the circuit to a USB charger (power only, no data), it boots normally.
But if I hit RESET, it stucks in the bootloader for a long time (dunno how long... more than 30sec, lazy to time it...).
However it finally runs my user program.

But, if I plug it into a computer (with USB data) and hit RESET, it does run pass to user program, every time.

And, with the supercapacitor in place, if I charge it to full capacity and hit RESET, it doesn't boot either.
If I plug it into a computer when it stuck, the computer does find the micronucleus bootloader, I can upload program to it.

I suspect that I pulled D+ to the 5v rail, and the input bypass capacitor for the LDO still holds a charge.
So I tried:
1. solder a 22k ohm resistor from 5V to ground, so it discharge the capacitors
2. modify the circuit a bit and pull-up with the 3v3 rail
3. doing it in software, set the D+ pin as OUTPUT and pull it LOW to discharge the capacitors.
however none of them works...

What possibly could I do wrong? do I have to pull D+ D- ground with something like 22k ohm or 47k ohm? if so, what are the proper value for the pull-down resistors?

cpldcpu, 2016-10-26:
I assume the ATtiny13 in your circuit is in fact an ATtiny85? 

So apparently somehow noise is injected into the D+ and D- lines when your device is not connected to the computer. This will extend the timeout delay. The newest version of MN is somewhat more robust to this issue - no more infinite loop. However, it is close to impossible to distinguish high frequency noise from an erronous USB transmission.

The best option to address this issue would be to reconfigure micronucleus so it is not always invoked at power up. For example by using the EXT_RESET entry option, or by checking for S1.

Palatis, 2016-10-26:
well yes, fritzing change t85 to t13 when I choose t85 from the menu. but they're pin compatible so i didn't really pay attention.
will pulling D+ / D- ground work? because if I leave USB unplugged, they're basically floating, so it's probably picking up some noise.

what would be the proper value for the pull-down resistor that's with minimal impact to power usage?

cpldcpu, 2016-10-26:
Pulling them up would be a better idea. But anyhow, the best option is not to invoke MN at all, if it is not needed, by using a proper entry condition.

Palatis, 2016-10-26:
so you mean that i also pull d+ up with 1k5？

cpldcpu, 2016-10-26:
Ah, right. Pull D+ down, D- up.

Palatis, 2016-10-27:
strange, I pulled D+ ground with 1k5 (cuz that's the one with highest resistance in 0402 size laying around...) but nothing really happens.

cpldcpu, 2017-08-20:
closing due to age

closed 2017-08-20 by cpldcpu