micronucleus/micronucleus #90: Trouble building micronucleus.exe for Windows

ShawnHymel, 2017-02-08:
I am trying to build the uploader tool for Windows after changing to a custom VID/PID in micronucleus_lib.h.  I have MinGW and MSYS installed. When I try to call "make" in the commandline directory, I get the following error:

```
$ make
Building library: micronucleus_lib...
gcc -I C:\MinGW\include -L C:\MinGW\lib -lusb -Ilibrary -O -g -D WIN -c library/micronucleus_lib.c
In file included from library/micronucleus_lib.c:30:0:
library/micronucleus_lib.h:35:82: fatal error: lusb0_usb.h: No such file or directory
compilation terminated.
make: *** [micronucleus_lib] Error 1
```

I downloaded both libusb-1.0.21 and libusb-win32-bin-1.2.6.0, but I am at a loss as to how to install them in MinGW such that gcc can find them. Any help is appreciated!

ShawnHymel, 2017-02-08:
I was able to construct a workaround. I copied libusb-win32 .h and .a files locally and changed the -I and -L parameters in the Makefile to point to the local copies.

closed 2017-02-08 by ShawnHymel

cpldcpu, 2017-02-09:
Glad you found the solution. Anyways, the lib and include path is noted in the build file, so you would alternatively have to copy header and lib there:

-I C:\MinGW\include -L C:\MinGW\lib