micronucleus/micronucleus #92: user program becomes unreachable sometimes

Genoil, 2017-04-28:
I have a custom PCB with (among other parts):
- attiny84A
- 12 Mhz crystal
- button (active high)
- LED (active high, driven from MCU)
- no RESET circuit/button
- on/off switch
- ISP header
- micro-USB port

I have a custom micronucleus configuration for this, based on the t84_default config. The main differences are pin assignments, and a change to the stock ENTRY_JUMPER code that allows using my button as a JUMPER to conditionally enter the bootloader when the button is held down while switching the device on:

```
#elif ENTRYMODE==ENTRY_JUMPER2
  #define bootLoaderInit()   {JUMPER_DDR&=~_BV(JUMPER_PIN);JUMPER_PORT&=~_BV(JUMPER_PIN);_delay_ms(1);}
  #define bootLoaderExit()   {JUMPER_PORT|=_BV(JUMPER_PIN);}
  #define bootLoaderStartCondition() ((JUMPER_INP&_BV(JUMPER_PIN)))
```

This works great! I can upload application firmware via USB and don't have to wait x seconds of bootloading timeout when he device is switched on by my end user. 

Now, the problem. I have 6 prototypes programmed and tested like this and now yesterday and today 2 of the boards (one of them twice) have gotten into a state where when you switch them on, the user application won't start, and when pressing the button _after_ the device has been switched on, it will suddenly enter the bootloader. This leads me to believe that micronucleus gets stuck somewhere in the startup process, but I don't know how to be sure and what to do about it. 

I have noticed that my LED faintly flashes a few times when getting into this state. Its seems like the DDR of the pin is still not set at this point, because the flashing is much less bright than when I press the button afterwards, when it enters the bootloader. It also won't directly enter bootloader mode when the button is pressed down while switching the device on, only afterwards.

Switching the device on/off will not get it out of this state, so it seems to be something permanent that tunrs up as the cause of _something_. When I re-flash the user program (or reflash the bootloader for that matter), it just all works fine again, but I can't send them out to my client in this state. 

My user app size was 99% of the max size before, but I've ran into the same issue with a shrunken down to 65% user app.

My fuse settings are: low:0xff high: 0xdf ext:0xfe.

cpldcpu, 2017-07-30:
This is the first time I hear of this issue. In principle, it is of course possible for a user program to corrupt itself. There is no write protection in the ATTiny, so it is impossible to prevent this.

closed 2017-08-20 by cpldcpu

AHorneffer, 2019-01-11:
Jaja, I'm late to the party... but I might still have some useful information. 

If I see that correctly, then you have the brown-out detection in the ATtiny disabled. I had a similar problem on an ATmega328p, sometimes so flash-pages got erased. After I got a tip on another forum and switched on the brown-out detection, this problem vanished. 

It is unsurprising that micro-controllers behave erratically when the voltage is too low, and it is apparently not too uncommon that they erase part of their flash when that happens.