micronucleus/micronucleus #94: USB timing

danjohn, 2017-05-10:
I have some DigiSpark (AT85) boards here, that worked well at the beginning.

After ~50 burn&crash cycles (thats a approved "development method"! ;-), things went bad:
It was harder recognized by my PC as an USB device from time to time.

I can only explain that as a timing drift with USB signals... does micronucleus store its timing parameters (AT85@16.5MHz) permanently somewhere? Or is the RC osc of my ATiny85 aging that fast?

Bluebie, 2017-05-10:
i think the oscillator would always recalibrate on boot. you can't depend on stored values because they are temperature dependant. maybe you could use an oscilloscope, if you have one around, to see if the signal on the data lines looks nice and crisp or more rounded off and weak, or alternately have some ringing (waving up and down after each digital change)? sometimes i've seen avr worsening communications troubles like that are sometimes from the line drivers in the chip overheating and wearing prematurely, if too much current is passing through for some reason, then the amount of current it can handle tends to go down gradually.

The only other thing i could think of is something to do with the temperature. I believe the RC oscillator only gets calibrated once on boot, so if the chip was really warm near the RC osc, and then was cooling down in the bootloader rapidly, maybe it could drift fast enough to throw it out sometimes?

danjohn, 2017-05-10:
Sorry, no Oszi by now.
I can only tell my experiences... the fresh boards worked well, but after programming them multiple times they produce more and more "device not accepting address, error -32; unable to enumerate USB device" errors when plugging them in.

I also recognized premature sign-offs using the DigiCDC lib, which also seems to rely on 1% timing constraints.
So I blame it to USB timing, but can't substancially nail down that conclusion...
Does "micronucleus" and "digiCDC" share some code in common?

All I know, is that my boards get more unrelieable the more times I plug them in -- so I thought, some timing parameters are calculated oddly, stored in Flash, and drifting away.
If both MN and DCDC "recalibrate on boot", then I'm clueless.

Thoughts, anyone?

Bluebie, 2017-05-10:
micronucleus and DigiCDC both use the V-USB software usb device library.

Alternate theory: the stupid pcb connector on digisparks wears out really quickly and gets dirty?

I had trouble with DigiUSB back in the day where it would stop working after a day or so. I don't know. These software usb things are a giant hack. It's kind of more surprising that they work at all, than that they aren't totally reliable.

GGreenwood, 2017-05-10:
If you're running the RC Oscillator at 16.5MHz, I'd recommend reading through the post [here](http://codeandlife.com/2012/02/22/v-usb-with-attiny45-attiny85-without-a-crystal/). In short, you can use the 1ms pulses from the host computer as a timing base to recalibrate your RC oscillator to during runtime.

If you wanted to make these changes to the Micronucleus bootloader, you'd need to make the modifications as described in the linked article.

It's also possible that the connectors are running thin, as @Bluebie mentioned. Try carefully retinning the data lines on the connector?

cpldcpu, 2017-05-11:
Well, in my experience USB is quite resilient. The protocol itself is able to cope with a few timing glitches. So I don't think that a slight clock inaccuracy will lead to noticable errors. And, as Bluebie pointed out above, the clock is recalibrated every time the device is plugged in again.

I would also wager that this is rather an issue with the PCB-based USB plug. These are definitely less of a good idea than it looks initially...

danjohn, 2017-05-11:
Ok, you blame the physical connenction (PCB stripes), but some of my boards got "real" mini-USB connectors (the cell phone charger type), so I won't bet on that. :)
Of course that contacts may wear out too, but as both board types show the same problem, I have to blame it to the firmware by now.

I now try to dig into /lib/modules/*/kernel/drivers/usb/ , but as I'm no kernel dev, I dont't expect too much enlightment... :(

danjohn, 2017-05-14:
Referencing my [investigation ad](https://github.com/micronucleus/micronucleus/issues/94#issuecomment-300741871), I have to admit that stumbling thru kernel code won't get my primary obsession. ;)
But maybe I found some helpful pieces for not-that-expert people coming across this thread:

*USB device detection is done via the usbcore driver (in kernel-space), thus enabling the detection of a USB device, even without having its specific driver ¹).*
I suppose, that's where the "device descriptor read/64, error -32" messages originate from.

A definition of that error numbers is given in ²), an interpretation (of error -32, -71) is tried in ³), which states:
*Error codes like -EPROTO, -EILSEQ and -EOVERFLOW normally indicate
hardware problems such as bad devices (including firmware).*

Contrary, the client side (µM, dCDC) says ⁴):
* Release 2008-02-16: Added optional routine usbMeasureFrameLength() for calibration of the internal RC oscillator.*
This release seems to be part of my "1.11" firmware.
I didn't yet dig into dCDC for what V-USB version and what extra code they used.

So thats for that. I was somewhat insecured when reading
*3.- EMI - Electromagnetic interference is also a big problem for V-USB* ⁵), as such a tiny digispark cannot be inserted into an USB port without accidentially touching live parts.

But all that can't explain why things where working at the beginning.
Well, I'm courious about your interpretations and helpful opinions!
*And now going to wash my dishes to temporary get away from this strangenesses. ;-)*

¹) <http://opensourceforu.com/2011/10/usb-drivers-in-linux-1/>
²) <file:///usr/include/asm-generic/errno-base.h>, <file:///usr/include/asm-generic/errno.h>
³) <http://elixir.free-electrons.com/linux/latest/source/Documentation/usb/error-codes.txt>
⁴) <https://github.com/micronucleus/micronucleus/blob/v1.11/firmware/usbdrv/Changelog.txt>
⁵) <http://vusb.wikidot.com/troubleshooting>

cpldcpu, 2017-05-20:
@danjohn these are all valid concerns. In the end, the only proper way of debugging this is to use a logic analyzer or storage scope to log the traffic. 

As bluebie pointed out, the clock in micronucleus is recalibrated every time the device is reset. Most user programs should do the same, but it could make sense to verify that is actually the case with your code.

danjohn, 2017-05-20:
Thank you @oldcpu @bluebie for your help (sorry, have no lab equipment yet)!
(As promised I looked into dCDC: There is calibration code inside, but I'm not sure if it is used in all configs. But, well, it's about µN here. :) )

The EMI issue is a true possibility, bcs after I desoldered any wires, and touched the device only wearing insulating gloves, one of my sparks registered as an USB member, at least. Not very practical indeed :-).

The others insist in doing "-32" errors.
Does the arduino ide give the possibility to accidenially overwrite parts of µN by uploading a sketch (5350 Bytes to a ATiny85)?

cpldcpu, 2017-08-20:
@danjohn at this time there is no known bug that allows overwriting micronucleus while uploading a user program. However, a malicious use program could easily do some damage as there is no write protection in the ATtiny.

cpldcpu, 2018-01-05:
closing, since there was no follow up.

closed 2018-01-05 by cpldcpu