micronucleus/micronucleus #96: Compiling and usage issue

SparkyCola, 2017-06-25:
Hi, I tried to run micronucleus on an ATtiny45. I could flash it with avrdude and Arduino ISP by copying and modifying the avr dude line from the Arduino IDE and modifying it.
However when running lsusb (after hooking up pin2 to d+ and pin3 to d- (pins of the chip, it's 3 and 4 in software)) i couldn't see anything new.
So I wanted to enable the led in micronucleus to see if was running ok, and after I modified the config and Makefile (I added `CFLAGS += -I /home/mattis/arduino-1.8.2/hardware/tools/avr/avr/include/`) in order to have inttypes.h and others. But when I run `make CONFIG=t45_default`i get: 
`/usr/lib/gcc/avr/4.9.2/../../../avr/bin/ld: cannot find -lm
/usr/lib/gcc/avr/4.9.2/../../../avr/bin/ld: cannot find -lc
/usr/lib/gcc/avr/4.9.2/../../../avr/bin/ld: cannot find -lattiny45
collect2: error: ld returned 1 exit status
Makefile:79: recipe for target 'main.bin' failed
make: *** [main.bin] Error 1`
I use Ubuntu 16.10 if that is of any help.

SparkyCola, 2017-06-25:
Here is my Makefile:
[Makefile.txt](https://github.com/micronucleus/micronucleus/files/1100527/Makefile.txt)
and my config:
[bootloaderconfig.h.txt](https://github.com/micronucleus/micronucleus/files/1100529/bootloaderconfig.h.txt)

cpldcpu, 2017-08-20:
Have you tried the supplied hexfile for attiny45?

cpldcpu, 2018-01-05:
closing due to no response.

closed 2018-01-05 by cpldcpu