micronucleus/micronucleus #104: Testing branch not found

mihaigalos, 2017-11-07:
Dear contributors,

I have adapted Micronucleus to offer beta/not yet tested support for AtMega328p, which is to be found in many arduinos. Reading the [Contributing.txt](https://github.com/micronucleus/micronucleus/blob/master/Contributing.txt), I have tried to push to [testing](https://github.com/micronucleus/micronucleus/tree/testing), using 

`(on local branch m328p)
git push -u https://github.com/micronucleus/micronucleus/tree/testing m328p`

I am receiving the following error message: 
> fatal: repository 'https://github.com/micronucleus/micronucleus/tree/testing/' not found

Since I had always pushed to origin, I figured I could substitute it for the link provided in the Contributing.txt ..

Can you please advise what are the steps for contributing? I would like to share my work and have a central point where to commit to.

Many thanks in advance,
Mihai

AHorneffer, 2017-11-08:
Hi

You cannot directly push (write) to somebody else's repository.  What you can - and should - do is to fork the repository in GitHub, push your changes to that repository, and then start a pull-request to have your changes applied to the root repository. See also: https://help.github.com/categories/collaborating-with-issues-and-pull-requests/

Unfortunately the maintainers of the micronucleus repository are not very responsive at the moment (or anymore?), but people like me will see your pull-request and then can pull your changes into their repositories and give your code a try.

cpldcpu, 2017-11-08:
Thanks for your comment - spot on.

An ATMega port is interesting. MN could probably be simplified quite a bit for the ATMEga, since they  offer bootpage proctection

closed 2019-01-11 by AHorneffer