micronucleus/micronucleus #105: Should we give @AHorneffer write access to the main micronucleus?

Bluebie, 2017-11-08:
@cpldcpu @ahorneffer

cpldcpu, 2017-11-08:
Surely some help would be useful if @Ahorneffer is willing to help out. I still get all the updates from github, but one problem is that I don't get around testing anymore. So I had to merge several contributions blindly. Some help would be greatly appreciated.

AHorneffer, 2017-11-08:
\*blush\*
I feel honored. But I don't really know about the core programming of microcontrollers, nor am I really set up to do testing. At the moment I have quite some time available but that will probably change in the not too far future.
I just went through the list of pull requests, and I probably would accept #87 and #102 mostly because I think they don't break anything, I could also look into #91, but #80 and #101 are beyond my expertise.

So, yes, I would be willing to help out. But don't expect much from me. (And to tell new users that they should do a fork and then a pull request doesn't require write access.)

Phoenix1747, 2017-11-08:
Hi, I'm somebody that nobody knows and here is my advice: @AHorneffer should have write access. With that being out of the way he could have a look at some different pull requests and do sort of a community management and issue review and stuff if he doesn't want to do the core programming.
*flies away*

NicoHood, 2017-11-08:
Its always better to have multiple members with write access in case someone dies. I've also added some personal friends as members just in case...

Some other projects suffer from that and then a fork needs to be started. The problem with that is, that the readme of the original project will never get an update and it takes time for everyone to switch.

So why not if you trust him. I was also added to another project because of that reason and because I contributed very much. In the end you can just continue to submit patches via PR. That would still make sense.

cpldcpu, 2018-01-05:
@Bluebie Did you grant write access to him? Can't see him in the list.

AHorneffer, 2018-01-05:
@cpldcpu I don't have write access, so: no. :-)

cpldcpu, 2018-01-06:
@Bluebie Can you give write access to him? I do not have the required permissions in this group.

Bluebie, 2018-01-06:
Invitation has been sent, and @cpldcpu should have admin access to the org now too.

AHorneffer, 2018-01-06:
Thanks. I guess we can close this now.

closed 2018-01-06 by AHorneffer

cpldcpu, 2018-01-06:
@Bluebie Thanks! 👍