micronucleus/micronucleus #108: Line 69 of upgrade.c looks suspicious to me.  And probably a NOP is missing, too

hilbix, 2017-11-19:
Sorry, but I am not familiar with ATiny yet (I do not own any, just looked into the code to understand if it is worthwhile, and yes, it looks very promising).  So I am not really sure, that my interpretation is correct.  But looking at https://electronics.stackexchange.com/a/205780 I get the impression, that the updater needs to be improved.

All I did is read the source, and my `/brain.exe` signalled a WeirdSemanticsException while parsing line 69 of following code:

https://github.com/micronucleus/micronucleus/blob/140cad6603187631b96ce2d9b791ea23c6e7a670/firmware/upgrade.c#L68-L71

A bit of googling later, I got the impression, that

https://github.com/micronucleus/micronucleus/blob/140cad6603187631b96ce2d9b791ea23c6e7a670/firmware/upgrade.c#L69

should read

        table[i] = 0xFFFF;

(please note the `i` showing up there now).  Right?  Having it 0xFFFF probably is better than

        table[i] = 0;  /* NOP */

because after erasing page 0 it is 0xFFFF anyway, which hopefully preserves one precious Flash reprogramming cycle.  0xFFFF seems to be an illegal opcode (according to Google, again), which skips the next instruction, or not.  Using `NOP` does not help much anyway, because we should be prepared for the case when the power goes out after erasing and before we have reprogrammed page 0.

So after reset, eventually page 1 will be reached, but we might skip over the first instruction there, due to the last instruction on page 0 is 0xFFFF.  So this instruction must not be important as well, right?!?  Hence, while we are at it, isn't there one single NOP missing in following, too?

https://github.com/micronucleus/micronucleus/blob/140cad6603187631b96ce2d9b791ea23c6e7a670/upgrade/upgrade.c#L172-L176

As this only fills out page 0 (provided that the program starts at offset 32), and if Murphy hits (a second time after power outage, as usual) on reboot, the first instruction on page 1 should be NOPped, too, right?  Hence this makes it 17 NOPs and not only 16.

PS: Please do not forget the important second rule of Murphyology:  To keep things properly in place, suspenders plus a waistbelt are not enough redundancy, once Murhpy hits (and remember:  According to the first rule, Murphy always hits).

corbolais, 2017-11-19:
Second that. Should read
   ` table[i] = 0xFFFF;`
Nice catch on the NOP count!

AHorneffer, 2017-11-20:
FYI: this issue is also addressed in pull-request: #101 Among quite a few other changes.

hilbix, 2017-11-24:
AFAICS except the NOP count.  But thanks, nice to hear that I am not the first one who noticed the table thing ;)

closed 2018-10-21 by cpldcpu