micronucleus/micronucleus #109: Attiny85 - Programmer doesn't work

nitbeat, 2017-11-29:
Hi,

I flashed the micronucleus bootloader (t85_default.hex) to the ATtiny85 using Arduino as ISP,
I used these fuses: 
-e -Uefuse:w:0xFF:m -Uhfuse:w:0xD7:m -Ulfuse:w:0xE2:m

**This is the programmer I have:**
[image]

After uploading the code using the Arduino & Digispark (Default - 16.5mhz)

**I get "100% complete" msg ->**

`Running Digispark Uploader...
Plug in device now... (will timeout in 60 seconds)
> Please plug in the device ... 
> Press CTRL+C to terminate the program.
> Device is found!
connecting: 16% complete
connecting: 22% complete
connecting: 28% complete
connecting: 33% complete
> Device has firmware version 2.3
> Device signature: 0x1e930b 
> Available space for user applications: 6522 bytes
> Suggested sleep time between sending pages: 7ms
> Whole page count: 102  page size: 64
> Erase function sleep duration: 714ms
parsing: 50% complete
> Erasing the memory ...
erasing: 55% complete
erasing: 60% complete
erasing: 65% complete
> Starting to upload ...
writing: 70% complete
writing: 75% complete
writing: 80% complete
> Starting the user app ...
running: 100% complete
>> Micronucleus done. Thank you!`

But when I try to test the code it doesn't work.

I seen this video and for this guy the USB device disapears after a few seconds, but in my case when I connect the programmer the USB device(Digispark bootloader) stays on all the the time.
https://www.youtube.com/watch?v=MmDBvgrYGZs

During the upload of the sketch I can hear the Device disconnect after 2 seconds into the flash and then reconnect again.

Any Idea what's wrong ? maybe my fuses aren't good ?

Thanks

AHorneffer, 2017-11-30:
Yes! Try setting the fuses to the ones from the configuration: `-U lfuse:w:0xe1:m -U hfuse:w:0xdd:m -U efuse:w:0xfe:m`

Actually, I'm surprised that the ATtiny manages to register as a USB device in the first place. If I see that correctly, the "-Ulfuse:w:0xE2:m" sets it to 8 MHz clock and not the 16.5 MHz clock that is needed. Also the "-Uefuse:w:0xFF:m" setting disables self-programming, so rendering the bootloader useless.

The fuse configuration for the t85_default is in: https://github.com/micronucleus/micronucleus/blob/master/firmware/configuration/t85_default/Makefile.inc It also contains some explanation, so I recommend to have a look at it.

nitbeat, 2017-11-30:
Thank you my friend, you clarified a lot here in your message 👍 

One more question, so the 16.5 MHz clock is for the flashing procedure (Digispark (Default - 16.5mhz))
but it won't affect the actual attiny chip ?
I would like the ATtinny to run in 8mhz

Thanks

AHorneffer, 2017-11-30:
You need to have the ATtinny running at 16.5 MHz for the bootloader to work, the USB timing depends on that. Once you wrote your program, you could change the fuse back to 8 MHz (and change it back to 16.5 when you want to change your program), but then you don't need micronucleus in the first place.

The USB code doesn't allow for clock speeds other than 12, 12.8, 15, 16, 16.5, 18, or 20 MHz. But if someone could modify it to run with the 8 MHz internal RC oscillator (from e.g. the ATmega328) I would be grateful. :-)
(See also: https://github.com/micronucleus/micronucleus/blob/master/firmware/configuration/t85_default/bootloaderconfig.h)

...

Ah, there is dedicated assembler code for each frequency in: https://github.com/micronucleus/micronucleus/tree/master/firmware/usbdrv/

nitbeat, 2017-11-30:
Sorry, I'm a little confused, so after I flashed the code using this programmer I need to connect is to my Arduino as ISP and then burn bootloader with the 8mhz fuse ?

That would make this programmer method not as easy and convenient as I thought it would be.

BTW, the fuses you gave me worked with LED PIN 1 (on the programmer) it was blinking,
but when I tried to change it to LED PIN 4 and test it on the bread board the LED didn't blink.

It might be because of the USB+ USB- that uses pin 4, I will try another pin output.

But still not sure about the clock speed
are you saying the programmer has external oscillator ? how does it work after flashing the code

AHorneffer, 2017-11-30:
Well, in short I wanted to say: If you want to use micronucleus, you need to use the 16.5 MHz!

Using the USB pins for IO of the ATtiny is iffy, you need to be careful there! [...] If you get it wrong, you might pull 50 mA out of the data pins from you computer. I'm not sure if it will like that...

When you program with the external programmer, then the ATtiny gets put into a special "programming mode" (reset gets pulled to low for the duration(\*)) so I guess it just ignores the fuse settings. It also makes the ATtiny behave more like a "dumb" flash drive and less like a micro-controller. In contrast micronucleus is a "normal" program that writes to the internal flash, so all the fuse settings apply (including the "normal programs are not allowed to write to flash" fuse). 
(\*) Which explains that once you set the fuse to use the reset pin for "normal" IO, then you cannot program it with an external programmer anymore.

nitbeat, 2017-11-30:
I understand, I learned my lesson, going back to my Arduino As ISP, and I will build a shield for that.
When it could work on 8mhz I will use it again :)

Thanks for your help buddy !

cpldcpu, 2018-01-05:
Closing issue, as this was solved?

closed 2018-01-05 by cpldcpu

nitbeat, 2018-01-05:
Yes, thanks