micronucleus/micronucleus #110: Share usbdrv between bootloader and userspace

thielj, 2017-12-04:
Hi, not sure if I got this right, but if I use Micronucleus and e.g. DigiCDC, I basically end up with two copies of the V-USB driver, right? Given the limited space on the ATtiny85, wouldn't it make sense to allow sharing some of that code? Or config options to provide e.g. a minimalistic debug serial put_char() function to userspace?

thielj, 2017-12-04:
Looks like @plasticassius had a similar idea:
https://github.com/micronucleus/micronucleus/pull/80#issuecomment-250286881

thielj, 2017-12-05:
Haha, sorry for suggesting the obvious. I personally wasn't aware until I ran out of space and started looking into the micronucleus source...

@Adam5Wu's code looks useable, but I would probably sacrifice a few bytes and freeze the "binary API" (exports table and function prototypes) or only allow them to vary with ENABLE_UNSAFE_OPTIMIZATIONS defined.

I'm still trying to understand the implications of the interrupt-free approach and how to plug in my other code. In particular, some guidance on how to use:

    bool (*f_beforePoll)(uint16_t rem_us, uchar dev_addr),   // AKA communicate()
    bool (*f_afterPoll)(uint16_t rem_us, uchar dev_addr)     // AKA checkpoint()

I'm looking to transmit data to the host and "at least one callback per 5ms" won't be sufficient to e.g. record the timings of RF or IR signals.

The relevant documents I found are here:
https://cpldcpu.wordpress.com/2014/03/02/interrupt-free-v-usb/
https://github.com/Adam5Wu/micronucleus/blob/USBExport/USBExport.md
https://github.com/Adam5Wu/micronucleus/blob/USBExport/examples/DigisparkMouseLite/USBExport.md

AHorneffer, 2017-12-06:
One point is, that @Adam5Wu's approach doesn't just use (part of) the V-USB code in the bootloader, but re-uses also the USB-loop of the bootloader. Because the bootloader doesn't have much to do while it is waiting for commands from the USB it is fine for it to spend most of its time in a busy-loop waiting for USB traffic. 

If I see it correctly, then `f_beforePoll()` is called after USB traffic was detected & received (or 5ms timed out) but before an answer is sent. Or made ready to be sent? I don't really understand if `usbPollLite()` sends an answer, or just make it ready to be sent with the next request. `f_afterPoll()` is called after the answer is sent or made ready. And if the host is waiting for an answer, then the execution time of both of them together may not exceed the time it takes the host to time-out and reset the USB connection.

For your application it might be better to use the ".exports" segment method to expose some (all) of the low-level V-USB code but keep the higher level handling in the application code. You won't save as much program space, but would get a more flexible USB implementation.
(And for really time-critical applications you probably need dedicated USB hardware anyhow. V-USB is a bit of a hack.)

Adam5Wu, 2017-12-07:
To my understanding, the philosophy of interrupt free USB is that interrupts are too slow to acknowledge host request and would cause too much retransmission. The MCU should dedicate most of the time watching the wire. Everything else are just "auxiliary" work.

My plan is to create a HID USB UPS from ATTiny, so this scheme fits my need - it occasionally takes voltage current readings, and report back to the host.

If other time critical work is required, as @AHorneffer said, probably it is the best to have a dedicated hardware to take care of the USB. A servant cannot serve two masters at the same time... :D

AHorneffer, 2017-12-07:
Hi @Adam5Wu thanks for joining. I really appreciate your work.

Was my explanation of `f_beforePoll()` and `f_afterPoll()` in the previous post correct? 

Do you have any comments about my idea of exposing the more low-level V-USB code? Do you know of any show-stoppers or pitfalls?

> To my understanding, the philosophy of interrupt free USB is that interrupts are too slow to acknowledge host request and would cause too much retransmission. 

I understood it, that the problem with interrupt based V-USB are not real retransmissions, i.e where actual data is exchanged, but it didn't arrive correctly. But that host asks for data which isn't ready yet and the MCU spends all its time in the interrupt handler receiving and deferring (sending NACKs) the host's request so that it doesn't have time to actually prepare the data. The interrupt free approach mitigates this by intentionally ignoring some of the host's requests and spending that time more productively.

> My plan is to create a HID USB UPS from ATTiny, so this scheme fits my need - it occasionally takes voltage current readings, and report back to the host.

I guess that'll work fine.

Adam5Wu, 2017-12-07:
@AHorneffer my bad, after re-reading the original blog, I think you are correct about the problem interrupt free was trying to solve.

I was mainly speaking from my operational experience that, if the workload inserted between two polls are too large, the interrupt-free version very quickly deteriorates to the point the device is no longer stable and just drop out on the host.

In a sense, either interrupted or interrupt-free, ATTiny seems must dedicate most of its time to attend the host, probably because it is quite under-powered.

The interrupt-free performs better because it intentionally ignores some request so to squeezes some extra time for itself. This achieves better real-work/protocol-work ratio. But it is somewhat dangerous, like a poor man driving a car with only one working break...

If you want interrupt-free in the user program, I would recommend do something similar I did, because there is really not much room left, push the timing even a little more, the protocol will collapse.

If you want interrupted version, then you will need to prepare two versions of some core functions modified in interrupt-free, this probably would increase the bootloader by another 500 bytes. The user program will have more flexible scheduling, but much slower transfer...

Adam5Wu, 2017-12-07:
I haven't touched the code for a long time. If I recall, PollLite seems to perform receive and send in one go. So the before poll callback give you a chance to schedule any send immediately, while after poll callback let you process what is received, and possibly queue something to send in next poll.

AHorneffer, 2017-12-08:
@Adam5Wu I believe it is an inherent "feature" of V-USB, that the MCU spends a **long** time just talking to the host. USB low-speed isn't that fast and whenever anything is happening on the bus the MCU is busy. But, hey, it's good enough for many applications.

I may try to adapt your implementation to the current version of micronucleus, but it is not high on my list so don't bet on it.

@thielj I believe there are some ideas for you on how to proceed.

thielj, 2017-12-08:
Looks like my previous reply was lost... Some more thoughts

(1) The MCU spents most of the time bit-banging the USB protocol, even processing packets for other devices. Emulating USB low speed is already a challenge, high speed impossible. As such, eliminating any overhead from that and having the loop in control is probably a good idea performance wise.

(2) I think the issue reported in the blog with the host being busy sending NACKs was fixed mainly due to an improved timing strategy.

(3) I've read something about ~1200 time-critical cycles to handle a single USB packet (~75us on 16.5MHz) and there can be multiple successive packets. Unless there is a way to let a callback steal a few cycles on a regular basis, this basically excludes any other near real-time activity.

(4) Adding extra callbacks to steal cycles isn't a good solution either. In user code, I probably want to use an interrupt enabled V-USB stack instead, and have callbacks whenever an extended amount of time is available to process my data.

(5) Given that micronucleus typically doesn't do much more than idle 5 seconds: do I really care about it's improved performance? No. I would rather maximize the available flash memory! A backport to the original interrupt-driven V-USB code might be more suitable for a micronucleus version sharing the V-USB stack.

(6) There are multiple issues with emulating CDC devices. The specs don't allow for bulk endpoints on a slow speed device and recent Windows (and Mac?) drivers reject them. As much I would love to have a handy USB debug serial on e.g. the Attiny85, I would rather implement a non-USB soft serial now and hook it up separately when necessary.

(7) Given the few remaining use cases (HID, or with dedicated USB drivers slow data acquisition and simple switching): should I really bother or go straight to ATmegaXXu2/4 chips?

plasticassius, 2017-12-13:
Hi, it's been a long time since I've thought about this, but I'd like to add a bit to this discussion:

It is true that servicing the USB port takes much of the Attiny's bandwidth, and that includes traffic due to other devices. That said, if there are not too many packets, it can work well.

A polling strategy by itself does not optimize throughput or ignore irrelevant packets, optimizations can be made with or without interrupts. Rather, micronucleus uses polling in order to make the interrupt vector freely available to clients that it loads. That way, the client can do what it wants with the interrupt.

Timing for servicing USB is much easier to satisfy with an interrupt routine than with a polling strategy. As long as client code keeps interrupts enabled, it doesn't need to poll the USB loop for "somewhat less than 50ms" according to v-usb documentation. In practice, the client can avoid polling for longer than that if it is able to accept some dropped packets. In a purely polling strategy, the packet must start to be received within about 20us (us, not ms) which doesn't allow for much. However, it should be possible to use pure polling in microneucleus and an interrupt in the client code that calls back in to receive the packet.

An alternative approach is to use a different loader than microneucleus which would make the entire flash available to the client.

cpldcpu, 2018-01-05:
To get back to the initial question: The V-USB version in micronucleus has been seriously stripped down and modified. Sharing some function with the userspece programe would mean that a lot of code would have to be re-added. While I agree that this could reduced overall codesize, it somehow defeats the idea of having a minimally-sized bootloader. Also, there are numerous hardcoded options in V-USB, it is very likely that something would break and fail to work when functions are sharing.

So, generally I would recommend against it.

Adam5Wu, 2018-01-05:
I had a similar idea of that of @plasticassius, which may make the V-USB implementation a little more generic.

* It seems the main speed up of interrupt-free comes from intensionally not acknowleding some host request and thus saving more CPU time for useful work
* The current implementation achieve this by disabling ISR and get user code to vulnetaraily poll
* **How about re-enable the ISR, but service the host request selective?**
    * If the data is not ready, the ISR can choose to quickly return without bitbang the acknowledgement
    * Of course, it has to keep an internal counter, so that it do occasionally send acknowledgement to avoid breaking the protocol and cause host to rest connection.
    * In this way, we achieved both gain more useful CPU cycles, and leave interrupt enabled so user code is easier to write.

cpldcpu, 2018-01-05:
Actually the main reason for "interrupt-less" V-USB was not speed, that was a nice side effect. The motivation was:

 * Obliviate the need of intererrupt vector patching, which could cause user programs to malfunction.
 * Reduce code size

Therefore going back to the ISR-version does not make too much sense.

Adam5Wu, 2018-01-05:
@cpldcpu: I see your point.

Imagine if we are able to share most of USB handling logic between bootloader and user code, then even if we increased the bootloader size by a couple of hundred bytes, we still end with net saving, right?

Meanwhile, because we have shared the USB handling, there is no need to patch the vector anymore, so the first benefit is kept as well.

And if we are able to get a faster speed, wouldn't that make everything better? :D

cpldcpu, 2018-01-06:
@Adam5Wu 

Sure, but that would be a completely different approach that is only beneficial to a small number of programs. So I it would not make sense to chose this approach for the baseline version.

However, there is no reason not to have a this as an option or a fork? Go for it :) Porting Micronucleus back to the normal V-USB should not be that difficult. The tricky thing will be to create a robust way to reuse the driver.

NicoHood, 2018-01-06:
That is all a "nice to have but not really required" feature. If you need more space, dont use a bootloader and an ISP instead. Or just use another MCU with real usb support and more flash. I understand that it is fun to do it just because you can, but this is not helpful for the majority of the micronucleus users.

cpldcpu, 2018-01-06:
Btw - if anyone wants to have some fun hacking the V-USB driver: There is still quite a bit of overhead left. More bytes could be saved by stripping the driver down even more, or even going for a full assembler implementation. 

See u-wire: https://github.com/cpldcpu/u-wire

thielj, 2018-01-07:
> Sure, but that would be a completely different approach that is only beneficial to a small number of programs. So I it would not make sense to chose this approach for the baseline version.

I disagree. Most people who use a V-USB bootloader would want to use at least some V-USB functionality in userspace (like a debug serial). Those who need the few hundred extra bytes that a shared interface would occupy are probably familiar with ISP anyway and able to update or remove the bootloader.

ArminJo, 2021-02-04:
@thielj You are invited to deliver a V-USB bootloader with the functions you require and a documentation how to acces it in a user program. We can add it to the existing bootloaders.

closed 2021-02-04 by ArminJo