micronucleus/micronucleus #111: MCUSR cleared - no information about reset source

thielj, 2018-01-04:
Hi

MCUSR is being cleared by Micronucleus, probably to avoid a fused watchdog causing an infinite reset loop. However, that means information about the reset source is lost. Would it be possible to backup the register somehow?

cpldcpu, 2018-01-05:
Hi,

the MCUSR is only cleared when the bootloader is entered. If the bootloader is not activated, the MCUSR is not touched. 

https://github.com/micronucleus/micronucleus/blob/master/firmware/main.c#L256

The easiest way to address your problem would probably be to define an entrymode that will not activate the bootloader every time. Otherwise you could simply change the source to not clear the MCUSR, with the risk outlined by you.

AHorneffer, 2018-01-05:
I guess you could also store the value similar to the way it is done with OSCCAL when the `OSCCAL_RESTORE_DEFAULT` option is set. (E.g. in `bootLoaderInit()` and `bootLoaderExit()`)

thielj, 2018-01-07:
Thanks, I can easily workaround this with a custom start condition.

The current implementation doesn't make much sense though. I may be wrong, but the one and only reason to clear MCUSR is because WDRF would force WDE to be on, preventing the watchdog interrupt from being disabled or handled in software. As Micronucleus is doing neither, there is no reason to clear WDRF. Or any of the other flags. See e.g. page 46 here [1].

Also, someone fusing the WDT is probably not interested in Micronucleus delaying the reset. At least I would prefer my user code to be executed ASAP when the WDRF flag set. Same for fusing brown out detection and BORF.

The only exception would be (ab)using the WDT to force a reset to enter the bootloader. My understanding is that I can handle this situation with a direct jump into the bootloader code instead?

[1] http://ww1.microchip.com/downloads/en/DeviceDoc/Atmel-2586-AVR-8-bit-Microcontroller-ATtiny25-ATtiny45-ATtiny85_Datasheet.pdf

cpldcpu, 2018-01-07:
Yes, this is a precaution to allow MN to work with fused watchdog. There is also a WDR operation in the mainloop:

https://github.com/micronucleus/micronucleus/blob/master/firmware/main.c#L297

ArminJo, 2018-12-07:
Hello,
I propose to use `MCUSR = ~_BV(WDRF);` instead of `MCUSR = 0;`
in line 202 + 206 of main.c. This increases code size only by 2.

it is decribed in my ATtiny85 Datasheet on page 44:
```
; Clear WDRF in MCUSR
ldi r16, (0<<WDRF)
out MCUSR, r16
```
This is what you really want to do, but leaves all the other flags unaffected.
I tested it with an ATtiny 85 and it worked!

cpldcpu, 2018-12-08:
Actually the example above writes 0x00 into the MCUSR, so it would be no change.

MCUSR&=~_BV(WDRF) would increase the code size by 4, since the MCUSR is not accessible with CBI/SBI.

ArminJo, 2018-12-08:
Ok you are right!
I referenceded to `bit is reset by a Power-on Reset, or by writing a logic zero to the
flag.` and I testet it myself and it worked this way as described.

And the size figure is:
```
out	0x34, r1	; 52
```
compared to:
```
ldi	r25, 0xF7	; 247
out	0x34, r25	; 52
```
which gives 2 bytes and is consistent to these figures:
```
   text    data     bss     dec     hex filename
   1596       0      44    1640     668 main.bin
```
and
```
   text    data     bss     dec     hex filename
   1598       0      44    1642     66a main.bin
```

ArminJo, 2019-11-22:
### This feature is now successfully implemented in my [fork](https://github.com/ArminJo/micronucleus-firmware)

ArminJo, 2021-01-26:
MCUSR can be accessed in the following way:
```
void setup() {
    uint8_t tMCUSRStored;
    if (MCUSR != 0) {
        tMCUSRStored = MCUSR; // content of MCUSR register at startup
        MCUSR = 0;// to prepare for next boot.
    } else {
        tMCUSRStored = GPIOR0; // Micronucleus puts a copy here
        GPIOR0 = 0;// Clear it to detect a jmp 0
    }

```

closed 2021-01-26 by ArminJo