micronucleus/micronucleus #113: Idea: port Micronucleus to STM8

cpldcpu, 2018-01-06:
Here is an idea I have been toying with for a while: Why not port Micronucleus to the STM8? The protocol is device insensitive, so it should work.

There are plenty of really low cost boards with STM8F103 and USB connector out there, which can be ordered for less that USD1.00, shipped.

The STM8 is a somewhat unusual MCU, but it should be fast enought to handle bit-banged USB. In fact, there already is a software-USB implementation for it:

https://github.com/rikka0w0/STM8-VUSB

cpldcpu, 2018-01-06:
More information on STM8 USB:

https://hackaday.io/project/20431-hid-multimedia-dial/log/56461-debugging-stm8-vusb-code

thielj, 2018-01-07:
I've done some optimized bit-banging in assembly and the timings on the STM8 were really hard to predict due to the pipelined execution. They would also vary depending on instruction alignment.

I do like the hardware, but developer tools are lacking behind. No gcc, no 'proper' inline assembler in the IAR tools. The only 'open' option was SDCC. 

Also, cheap boards often use the internal RC instead of a crystal.

closed 2020-08-23 by cpldcpu