micronucleus/micronucleus #114: Having problem using windows command line tool

eeyrw, 2018-01-10:
I have installed driver for micronucleus device and it appears in device tree in device manager within short 6 seconds. But when I run pre-built micronucleus.exe. it shows that libusb-0-1-4.dll is required. So I copy a libusb0.dll and rename it accordingly. Then the tool seems to work but unfortunately it never senses the boot-loader device... Furthermore, I check previous commit and issue about libusb library and I discover that a version which use newest libusb library exists but it seems have compatible problem under Windows 10. So which driver should I install and how to make it work under windows 10？

jolabpi, 2018-01-16:
Same issue here but copying libusb9.dll and renaming it doesn't work for me. Micronucleus.exe cannot start.

eeyrw, 2018-01-17:
I downloaded a zip file from https://github.com/digistump/DigistumpArduino/releases/download/1.6.7/Digistump.Drivers.zip and unzipped it and get a pre-compiled micronucleus.exe. It works for me.

jabermony, 2018-02-04:
I've been bumbling around getting this working for me and I thought I'd leave a note for anyone in a similar situation until this is resolved more officially.
I ended up checking out the d94a1bdd26a5ed0ebec3532715d406153f7b5811 revision and using the pre-compiled version from there.
Your mileage may vary but I found that to work well in a windows 10 environment despite the issues you mentioned.
Likely related to my success with this method was that I installed the drivers under Zadig as WinUSB instead of the other options (e.g. libusb-win32) afaik WinUSB is a windows variant of libusb 1.0 where libusb-win32 supports libusb 0.1 on windows. Perhaps @eeyrw 's version from the digistump drivers zip was for libusb-win32 (or libsusb 0.1 compatible) whereas the one I've dug up is 1.0 compatible.
I'm by no means an expert on these things but hopefully that'll be useful for someone with the same issues.

Tvde1, 2018-07-07:
I'm getting the same issue. I installed the driver but no luck.

cpldcpu, 2018-10-21:
Curiously I encountered similar problems after reinstalling windows. I recompiled the commandline exe with a fresh install of mingw32 and libusb-win32 with static libraries. Please let me know if it works now.

closed 2018-12-10 by cpldcpu

djmips, 2019-09-18:
From what I can tell the 2.04 version has reintroduced this problem. I can use the version you built in Oct 2018 but not the latest 2.04 which fails with the libusb-0-1-4.dll dependency not found. On Windows 10 laptop.