micronucleus/micronucleus #115: udev rules for Digispark little wire shield...

jeanrocco, 2018-02-01:
The Digispark with little wire shield changes it's idVendor (16d0) and idProduct (0753) to 1781 and 0c9f respectively when plugged in, on my Linux Mint 18.2 Cinnamon 64-bit. So I added a new udev rule 49a-micronucleus.rules with this content:
SUBSYSTEMS=="usb", ATTRS{idVendor}=="1781", ATTRS{idProduct}=="0c9f", MODE:="0666", GROUP="dialout", OWNER="myuserid"
KERNEL=="ttyACM*", ATTRS{idVendor}=="1781", ATTRS{idProduct}=="0c9f", MODE:="0666", ENV{ID_MM_DEVICE_IGNORE}="1", GROUP="dialout", OWNER="myuserid"

Note that I had to add OWNER="myuserid" to get it working with Arduino IDE in my user space, I don"t have to run Arduino as root that way.
Hope this help, thank's
jeanrocco

nonchip, 2018-09-12:
seems kinda unrelated to micronucleus though, maybe you want to tell that to the digispark/littlewire people? also you didn't have to add the owner, you just have to add your account to the dialout group.

jeanrocco, 2018-09-12:
Hello Kyra, I'm happy you noticed this, and took the time to tell me. My
goal was to tell my udev experience with  devices that use micronucleus...
I would have liked to find in here such an info at the time ! This is also
for Arduino people, who are not necessarily linux gurus...
It all started with the comment ( https://disqus.com/by/jeanrochblais/ )  I
made on http://weininger.net/how-to-write-udev-rules-for-usb-devices.html
excellent write-up. So I told myself why not spread the good words :-) !
even for micronucleus users ...
As for the "owner", an id command shows me I'm indeed part of the dialout
group, but I still have to add it in the udev rules for it to work ... I
don't remember how far I tested that, but wth it works !
Cheers
jrb.

nonchip, 2018-09-13:
well you said it applies to littlewire, which isn't related to micronucleus and i don't think too many will end up here if they have issues with littlewire, that's why i thought maybe also spread the word wherever that is maintained :)

after looking up your comment i noticed it looks like you didn't actually show us the parts of the ruleset that apply to micronucleus in your case, adding those to your post above might be helpful

about the ownership: the only ways i can explain that is the arduino IDE being weird about it (wouldn't surprise me, arduino isn't really known for being the greatest IDE :P), or Mint doing something wrong with e.g. selinux or something like that. because those `0666` permissions clearly should not be any different between the `6` for the user and the `6` for the group. you found some REALLY strange behaviour there :P

jeanrocco, 2018-09-13:
hello kyra, it was nice chatting with you,
Our ramblings might be usefull after all... bye
jrb.