micronucleus/micronucleus #116: I want to write my firmware, but is some mistake

JIMhackKING, 2018-02-23:
I don't know how to make my firmware, I've create a new folder in firmware/configuration/, it is my new configure, I see here is a Makefile.inc file, so I use the make tool in firmware/ (I do not know if this method is correct) , but the terminal give me an error

`process_begin: CreateProcess(NULL, avr-gcc -I. -g2 -Os -Iconfiguration/t85_defau
lt -mmcu=attiny85 -DF_CPU=16500000 -DBOOTLOADER_ADDRESS=0x1980 -nostartfiles -ff
unction-sections -fdata-sections -fpack-struct -fno-inline-small-functions -fno-
move-loop-invariants -fno-tree-scev-cprop -x assembler-with-cpp -c crt1.S -o crt
1.o, ...) failed.
make (e=2):
Makefile:49: recipe for target 'crt1.o' failed
mingw32-make: *** [crt1.o] Error 2`

How to solve this problem? thanks. I can not find any problems or articles about compiling the firmware。

cpldcpu, 2018-02-24:
You need avr-gcc including binutils to build micronucleus. Have you verified with another project that your setup works?

JIMhackKING, 2018-02-24:
I haven't. How to include the things you say? I use Atmel studio and CodeVisionAVR, the burning software is progisp. My ability is not strong, so a lot of things do not understand, please give me a hank, thanks.

cpldcpu, 2018-02-24:
Well, as mentioned, the main thing is that you need the most recent version of avr-gcc including binutils. There are packes are for various operating systems. Everything else shoudl work out of the box.

closed 2018-07-28 by cpldcpu