micronucleus/micronucleus #123: User program won't start without a pull-up

bergie5737, 2018-06-11:
My issue is related to issue #86 which was closed due to its age.
I also found that as soon as the usb interface (the zeners and pull ups) is disconnected, my code won't start. My code works perfectly with v1.1. I've tried to change various settings in "usbconfig.h" and "bootloaderconfig.h" but as I am no good at programming, made no headway. 

When I leave the 1k pull-up resistor in place, my code runs. I hooked up my oscilloscope, and something triggers that pin to pull low during booting, it is not just "looking" for a high, it actively toggles the pin as well. Of course I have no idea how it is intended to work. I can only observe.

cpldcpu, 2018-07-28:
What kind of bootloader entry mode are you using?

ArminJo, 2018-12-06:
Hello cpldcpu,

I have the same issue as Bergie5737 above.
MN 1.11 works fine but 2.3 hangs forever, if the supply for the pullup at D- is detached. Attaching it manually few milliseconds after power up seems to cure it, but you have to be fast otherwise it has no effect.
So even attaching it "later" does not help, the user code is not entered.

My goal is to use a Digispark ATtiny85 board for a battery operated temperature alarm. Without the pullup, the modified Digispark needs 6 mico Ampere while sleeping, with the pullup it consumes 2.5 mA.
My modification is to connect the pullup directly at the 5Volt line of the USB, so it is not powered by the battery because of the diode between USB VCC and battery VCC. 
Programming works fine with these setup :-).

I guess my entry mode is the standard ENTRY_ALWAYS, since I took the loader from the release directory.
Is there any chance to fix these "feature" or am I stuck at the 1.11 version?

ArminJo, 2018-12-08:
Hi,
The reason for my problem is, that after the detection of a reset condition `calibrateOscillatorASM()` is called, which waits for the D- line to toggle and waits forever in my case.
My solution is to  go back to the 1.x versions behavior and call `calibrateOscillatorASM()` only if reset condition vanishes / SE0 State ends, otherwise check for timeout.
I changed part of the main() to
```
    uint8_t resetDetected = 0;
...
        if ((USBIN & USBMASK) !=0) {
            resetctr=100;
            if (resetDetected) {
                resetDetected = 0;
#if (OSCCAL_HAVE_XTAL == 0)           
                calibrateOscillatorASM();  // this waits also for reset condition to end 
#endif              
            }
        }
        if (--resetctr == 0) { // reset encountered
           usbNewDeviceAddr = 0;   // bits from the reset handling of usbpoll()
           usbDeviceAddr = 0;
           resetDetected = 1;
        }
```
this increases the code by 14 bytes.
In my fork, I introduced a START_WITHOUT_PULLUP define to save the 14 bytes of functionality if it is not needed.
see https://github.com/ArminJo/micronucleus/blob/master/firmware/main.c
Is this OK for you?
If yes, tell me than I can issue a pull request.

ArminJo, 2019-01-11:
I just created a new [branch ](https://github.com/ArminJo/micronucleus/tree/no_pullup) which contains only the solution to #123