micronucleus/micronucleus #129: Extra compactness for the bootloader

rradar, 2018-12-06:
Hello everyone!

I wanted to do a project ([ATtinyPowerMeter](https://github.com/moononournation/ATtinyPowerMeter/issues/1)) which is originally made with a attiny85. I want to replace it with a digispark I own but have troubles because the program code is to big.

I'm using the stock bootloader and wonder if there is any _extra light_ one were it would be possible to **save another 502 bytes for the program code**?

It would be so nice :yum: 

[image]

avaldebe, 2018-12-06:
Maybe you can save some programming space with a different OLED library.
In found the [Tinusaur OLED](https://github.com/Defragster/ssd1306xled) to be quite compact...

rradar, 2018-12-06:
Thanks @avaldebe, this was quick!

I also found out that there is a X-tra small firmware available...

> t85_aggressive  -   ATtiny85  smaller size - critical   -   1426 bytes

...which should just be enough for this project! I will see if I get it burned :fire: :stuck_out_tongue_closed_eyes:

rradar, 2018-12-06:
It worked out to update the boot loader but now I can't flash anymore program code via arduiono

see here: https://github.com/micronucleus/micronucleus/issues/130

closed 2018-12-06 by rradar