micronucleus/micronucleus #131: Digispark does not detected for code upload with micronucleus 2.3

rradar, 2018-12-06:
Last issue (https://github.com/micronucleus/micronucleus/issues/130) solved - new issue rising

So weird! :weary: My digispark does not get detected to upload code in the arduino IDE

I can plug it many times but it never gets anything uploaded :cry: 

```
Sketch uses 6514 bytes (97%) of program storage space. Maximum is 6714 bytes.
Global variables use 110 bytes of dynamic memory.
Running Digispark Uploader...
Plug in device now... (will timeout in 60 seconds)
> Please plug in the device ... 
> Press CTRL+C to terminate the program.
> Device search timed out
```

This is my dmesg:

> [60001.234567] usb 1-2: new low-speed USB device number 69 using xhci_hcd
> [60001.234568] usb 1-2: New USB device found, idVendor=16d0, idProduct=0753
> [60001.234569] usb 1-2: New USB device strings: Mfr=0, Product=0, SerialNumber=0

Any help highly appreciated! :fireworks:

rradar, 2018-12-06:
I wonder if it could be connected to this part in the instructable I used to upgrade the bootloader:

> [...]
> And if that gives us information about the fuses and the chip, we'll proceed with setting the right fuses...
> 
> `$ avrdude -p attiny85 -C avrdude.conf -c linuxgpio -v -U lfuse:w:0xe1:m -U hfuse:w:0xdd:m -U efuse:w:0xfe:m`
> This command will NOT disable the reset function on the DigiSpark.
> 
> The one bellow however, will! And don't use it just yet. You can use it at the very end, but you'll need HVSP programmer to re-enabe the reset function for future upgrades of the bootloader. On the other hand, you'll be able to use it as an SPI programmer using Little Wire...
> 
> `$ avrdude -p attiny85 -C avrdude.conf -c linuxgpio -v -U lfuse:w:0xe1:m -U hfuse:w:0x5d:m -U efuse:w:0xfe:m`

I didn't do this at the very end as I understand it's not mandatory. I don't have a need for pin 5 right now so I thought it's better to have a way back in case of emergency.

Also I think the bootloader itself works fine because it used to report:

> Device reports version as: 2.3

in arduino IDE before I solved this issue here: https://github.com/micronucleus/micronucleus/issues/130

rradar, 2018-12-07:
Finally get *something*
I added a old usb 2 hub in one of my usb 3 ports and tried again.
It does not work as expected but at least I get *some* output. Eventhough uploading fails :cry: 

```
Sketch uses 6514 bytes (97%) of program storage space. Maximum is 6714 bytes.
Global variables use 110 bytes of dynamic memory.
Running Digispark Uploader...
Plug in device now... (will timeout in 60 seconds)
> Please plug in the device ... 
> Press CTRL+C to terminate the program.
> Device is found!
connecting: 16% complete
connecting: 22% complete
connecting: 28% complete
connecting: 33% complete
> Device has firmware version 2.3
> Device signature: 0x1e930b 
> Available space for user applications: 6714 bytes
> Suggested sleep time between sending pages: 7ms
> Whole page count: 105  page size: 64
> Erase function sleep duration: 735ms
parsing: 50% complete
> Erasing the memory ...
erasing: 55% complete
erasing: 60% complete
erasing: 65% complete
> Starting to upload ...
>> Flash write error -1 has occured ...
>> Please unplug the device and restart the program.
```

Any ideas anyone?

cpldcpu, 2018-12-08:
Don't you have any non-USB3 port? Hubs can create lots of issues with Micronucleus and V-USB in general, especially the older type that forwards all transmissions.

rradar, 2018-12-08:
> Don't you have any non-USB3 port?

No. Only USB 3 Ports. I don't get any feedback on them. Just when I plug in the USB 2 hub in one of my USB 3 Port I get the output from my last message. I will try a different machine with a native USB 2 port and report back!

cpldcpu, 2018-12-08:
Well, you observe this kind of behavior of the USB communication is instable. One reason could be interference from other devices. 

I just noticed that you followed instructions to flash the "t85_agressive" firmware. This version is taking a few shortcuts to reduce code size. Did you also try the normal one?

rradar, 2018-12-08:
> I just noticed that you followed instructions to flash the "t85_agressive" firmware. This version is taking a few shortcuts to reduce code size. Did you also try the normal one?

No. Because I need the few bytes I'm saving for the program code. But right now I'm preparing another host with usb 2 ports to check if uploading with arduino works there. I'm reporting back soon

rradar, 2018-12-08:
Yes! It worked out on the other host with native USB 2 ports and some patience!

I going to close here. Troubles were probably caused by my USB 3 ports!

closed 2018-12-08 by rradar

AHorneffer, 2018-12-08:
@rradar 
FYI: It's not linked to USB 3 vs. USB 2. On my PC I have a few USB ports that work and others that don't work: the back USB 3 ports work, the back USB 2 ports don't work, and the front (USB 2) ports work most of the time...

rradar, 2018-12-08:
@AHorneffer 
for my notebook with three usb 3 ports it doesn't work on any of them. But I guess they all sharing one (shitty?) hub. For the other computer which just had two usb 2 ports it worked straight away on the first one (the other was occupied with a keyboard/mouse receiver)
I guess for bigger PCs with more than one hub (for example a computer with 6 usb ports would need at least 2 hubs in my world) it's a different chapter. Specially if you think one hub after another :man_facepalming:

rxaviers, 2021-07-01:
FWIW, in my case the problem was the USB cable.

spuder, 2022-08-28:
I know this is an old thread, but for anyone coming here from the internet the best workaround is 

1. Use a USB 2.0 hub
2. Use a USB Extension cable

Additional information in this [youtube video](https://www.youtube.com/watch?v=hVQubqgdoZg)