micronucleus/micronucleus #134: Disconnecting micronucleus leave USB -D pullup active

ArminJo, 2018-12-13:
In Issue #52 it look like it was fixex, but now I see
```
initHardware();  /* Disconnect micronucleus */
```
which ends with `usbDeviceConnect();` instead of
```
usbDeviceDisconnect(); /* Disconnect micronucleus */
```
as proposed in issue #52.
If you specify `USB_CFG_PULLUP_IOPORTNAME` then the port is still active powering the pullup after disconnecting micronucleus.

AHorneffer, 2018-12-19:
Hmmmm...
Where is `USB_CFG_PULLUP_IOPORTNAME` defined? As far as I can tell, that functionality of VUSB is not really supported by micronucleus.

I don't know when (if at all) I have time to look into this (it won't be before the new year), but if you think you have a solution, then you can set up a pull request.

ArminJo, 2019-01-02:
Hi,
I set up the pull request.
Happy new year
Armin

AHorneffer, 2019-01-11:
As far as I can tell, the change "back" to putting `initHardware()` into the code was in commit: 584f66ac8a88a554aae426b7112dd3254a08924e.

For a "normal" micronucleus (without `USB_CFG_PULLUP_IOPORTNAME`) the `usbDeviceDisconnect()` will pull the D- line low, and keep it pulled low if that is not reset. `initHardware()` will do this "_reset_" by calling `usbDeviceConnect()` after a delay. The delay alone costs 18 bytes of code, so having that in there twice is discouraged.

ArminJo, 2019-11-23:
### This feature is now successfully implemented in my [fork](https://github.com/ArminJo/micronucleus-firmware)

closed 2019-11-23 by ArminJo

ArminJo, 2020-05-19:
I deleted the according branch.