micronucleus/micronucleus #138: "USB Device not Recognised" after user program started

A-Ibrahim, 2019-01-10:
compiled latest firmware and flash it on ATtiny85 
Windows 10  popup "USB Device not Recognised" 
 micronucleus --run sketch_jan07a.ino.hex
> Please plug in the device ...
> Press CTRL+C to terminate the program.
> Device is found!
connecting: 16% complete
connecting: 22% complete
connecting: 28% complete
connecting: 33% complete
> Device has firmware version 2.3
> Device signature: 0x1e930b
> Available space for user applications: 6522 bytes
> Suggested sleep time between sending pages: 7ms
> Whole page count: 102  page size: 64
> Erase function sleep duration: 714ms
parsing: 50% complete
> Erasing the memory ...
erasing: 55% complete
erasing: 60% complete
erasing: 65% complete
> Starting to upload ...
writing: 70% complete
writing: 75% complete
writing: 80% complete
> Starting the user app ...
running: 100% complete
>> Micronucleus done. Thank you!
---------------------------
dmesg output in Linux
[  136.944601] usb 1-1: New USB device found, idVendor=16d0, idProduct=0753
[  136.944626] usb 1-1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[  141.798231] usb 1-1: USB disconnect, device number 2
[  142.417519] usb 1-1: new low-speed USB device number 3 using dwc2
[  142.637355] usb 1-1: device descriptor read/64, error -71
[  142.977137] usb 1-1: device descriptor read/64, error -71
[  143.316868] usb 1-1: new low-speed USB device number 4 using dwc2
[  143.536746] usb 1-1: device descriptor read/64, error -71
[  143.876548] usb 1-1: device descriptor read/64, error -71
[  143.996611] usb usb1-port1: attempt power cycle

AHorneffer, 2019-01-11:
This is a known behavior if the user program does not have USB functionality itself. (See the change notes for v2.01 in the `Readme.txt`.)

Is this an actual problem?

A-Ibrahim, 2019-01-12:
yes it is actual problem for me right now
- on windows 10 popup "USB Device not Recognized" and this does not happen if I downgrade to Micronucleus V1.11
- on Linux my USB port hang and never reconnect to any other device till I reboot my system
if I downgrade to Micronucleus V1.11 everything back to normal

AHorneffer, 2019-01-12:
> on Linux my USB port hang and never reconnect to any other device till I reboot my system
if I downgrade to Micronucleus V1.11 everything back to normal

Oh, I don't have this problem on my computer (running Ubuntu 16.04). I get more or less the same messages in my syslog, but when I pull out the device and plug it back in then it is recognized again.

FYI: My dmesg output.
```
[ 2053.301334] usb 1-10: new low-speed USB device number 4 using xhci_hcd
[ 2053.541237] usb 1-10: New USB device found, idVendor=16d0, idProduct=0753
[ 2053.541241] usb 1-10: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[ 2053.567309] usb 1-10: USB disconnect, device number 4
[ 2054.137568] usb 1-10: new low-speed USB device number 5 using xhci_hcd
[ 2054.293335] usb 1-10: device descriptor read/64, error -71
[ 2054.529317] usb 1-10: device descriptor read/64, error -71                                                             
[ 2054.765269] usb 1-10: new low-speed USB device number 6 using xhci_hcd
[ 2054.921493] usb 1-10: device descriptor read/64, error -71
[ 2055.157272] usb 1-10: device descriptor read/64, error -71
[ 2055.264957] usb usb1-port10: attempt power cycle
[ 2055.917144] usb 1-10: new low-speed USB device number 7 using xhci_hcd
[ 2055.944653] usb 1-10: Device not responding to setup address.
[ 2056.180950] usb 1-10: Device not responding to setup address.
[ 2056.389139] usb 1-10: device not accepting address 7, error -71
[ 2056.517080] usb 1-10: new low-speed USB device number 8 using xhci_hcd
[ 2056.544965] usb 1-10: Device not responding to setup address.
[ 2056.780974] usb 1-10: Device not responding to setup address.
[ 2056.988758] usb 1-10: device not accepting address 8, error -71
[ 2056.988827] usb usb1-port10: unable to enumerate USB device
[ 2072.335354] usb 1-10: new low-speed USB device number 9 using xhci_hcd
[ 2072.578942] usb 1-10: New USB device found, idVendor=16d0, idProduct=0753
[ 2072.578946] usb 1-10: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[ 2072.607666] usb 1-10: USB disconnect, device number 9
[ 2073.179273] usb 1-10: new low-speed USB device number 10 using xhci_hcd
[ 2073.335301] usb 1-10: device descriptor read/64, error -71
[ 2073.571275] usb 1-10: device descriptor read/64, error -71
[ 2073.807202] usb 1-10: new low-speed USB device number 11 using xhci_hcd
[ 2073.963236] usb 1-10: device descriptor read/64, error -71
[ 2074.199202] usb 1-10: device descriptor read/64, error -71
[ 2074.307197] usb usb1-port10: attempt power cycle
[ 2074.959077] usb 1-10: new low-speed USB device number 12 using xhci_hcd
[ 2074.986833] usb 1-10: Device not responding to setup address.
[ 2075.222805] usb 1-10: Device not responding to setup address.
[ 2075.431021] usb 1-10: device not accepting address 12, error -71
[ 2075.559008] usb 1-10: new low-speed USB device number 13 using xhci_hcd
[ 2075.586756] usb 1-10: Device not responding to setup address.
[ 2075.822729] usb 1-10: Device not responding to setup address.
[ 2076.030958] usb 1-10: device not accepting address 13, error -71
[ 2076.031027] usb usb1-port10: unable to enumerate USB device
```

AHorneffer, 2019-01-12:
You can change the behavior of micronucleus by changing the way it disconnects from the USB, with the change that was done here: https://github.com/ArminJo/micronucleus/commit/a5f798fa53c4f93e72da9fcae25bf920a04340da 
As I wrote in #134, that was intentionally changed in 584f66ac8a88a554aae426b7112dd3254a08924e. 

Or you could just kick out these lines altogether and have micronucleus not disconnect from the USB host in the first place. (This seems to be how micronucleus 1.x did things.)

My current understanding of the issue is, that the correct(TM) way to do this would require using a pin for the pullup on the D- line, so that you can make both USB data lines neutral (high impedance) like you physically disconnected the device.

A-Ibrahim, 2019-01-12:
I made the change in source code as suggested "usbDeviceDisconnect();" in main.c 
it worked for me quite well both problems  Windows 10 & Linux disappeared.

closed 2021-02-04 by ArminJo

maulikg78, 2021-08-23:
Hi, am late to the party, but can someone please help? I downloaded the digispark driver for windows 10 from following link and am getting the above error 

https://github.com/digistump/DigistumpArduino

It starts by showing digispark board connected under 

ibusb-win32 Devices, digispark bootloader 

but immediately after that it gets disconnected mentioning "USB device has malfunctioned in pask, windows in unable to recognize the device"

ArminJo, 2021-08-24:
This is expected, since the USB client is only active for a few seconds, dependent on the mironucleus kernel burned on the device.