micronucleus/micronucleus #141: Micronucleus support in Arduino IDE

Harvie, 2019-01-15:
Currently i use Attiny85 with micronucleus and in arduino IDE i have to add "digispark" board repository:
`http://digistump.com/package_digistump_index.json`. But they only support ATTiny85 and ATTiny167. There is no support for other uCs supported by micronucleus. Eg.: digispark arduino package has no support for ATMega328p which is supported by micronucleus.

I think it would be useful if micronucleus provided similar repository, that i could import to Arduino IDE to get full support for any device running micronucleus. Without using 3rd party repository.

per1234, 2019-01-15:
Here's the Arduino hardware package specification:
https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5-3rd-party-Hardware-specification

If you want to add support for easy installation via Boards Manager (shown in the screenshots above), here is the documentation for that:
https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.6.x-package_index.json-format-specification

Please let me know if you have any questions.

cpldcpu, 2019-01-15:
Hm... I wonder how that would work logistically. Don't you also need to indicate pin locations for each new board? For example, I don't think it would be enough to create one generic board for Micronucleus ATMega328 as there could be boards with different USB pins.

Harvie, 2019-01-15:
Another feature that Arduino IDE provides is called "burn bootloader", so this could be one way to distribute micronucleus. If you provide bootloader as part of this package, user can simply select MCU and ISP programmer from arduino IDE and by single click he can flash the bootloader.

Also you can provide board-specific example sketches in such package.

AHorneffer, 2019-01-16:
@Harvie: I might look into this. But don't know when...

@cpldcpu: The Arduino pin-numbers ("digital pin 1" etc.) are tied to the chip. So it's the task of the hardware developer to label the ports on an experimenting board in an Arduino-compatible fashion. For just loading the application program you don't actually need to know which pins are used for the USB, as long as the hardware and the bootloader in the chip agree. The main convenience gain for the user would be that the micronucleus commandline tool could be started directly from the Arduino IDE.
For burning the bootloader into a fresh chip, my suggestion would be to provide a number of example bootloader files and document which pins they use for the USB. Newbies are usually happy when they get told which pins to connect where, and more advanced users should be able to compile their own firmware and get that loaded onto the chips.

Harvie, 2019-01-17:
> The Arduino pin-numbers ("digital pin 1" etc.) are tied to the chip.

I think that he was talking about V-USB usbconfig.h, which needs to specify pins used for USB communication. But there are only few commonly used configurations, so this should not be a big deal.

For arduino pin numbers (eg. when using digitalwrite) at attinys i think it's good enough to map them to the DIP8 pin number. As a bonus you can add identifiers such as PB1 for first bit of PORTB, etc...

cpldcpu, 2019-01-17:
>I think that he was talking about V-USB usbconfig.h, which needs to specify pins used for USB communication. But there are only few commonly used configurations, so this should not be a big deal.

Yes, indeed. That was what I was referring to. Well, if there are only few combinations it should be fine.

I think it is a good idea to provide Arduino hardware packages. We could host them on this Github group, similar to how it is done for the Digispark. Should we create a new repository, though to avoid overloading the micronucleus one with megabyte sized datablobs?

Harvie, 2019-01-18:
> Well, if there are only few combinations it should be fine.

I guess that you can have lots of combinations of pins, but there are some reasons why people don't do that. Namely:

 * Most of libraries/bootloaders have some defaults. I've used micronucleus defaults and as a beginner it didn't make any sense to me to reinvent the wheel and try something other than defaults. It was just working and i had no need to change that so far.
 * People are just copying the schematics from each other. I see some blogpost of somebody who has v-usb working, i copy his wiring and make a new blogpost about my setup and somebody copies that from me.
 * There are some practical limitations. Eg. you might want not to use v-usb on pins/interrupts that have some other usefull features like PWM or ADC, so you can use these for another stuff.

So most people end up using more or less the same setup. If you include few of such reccomended setups in the arduino package, it will make even more people use that and therefore further improve the compatibility in the long run :)

> Should we create a new repository, though to avoid overloading the micronucleus one with megabyte sized datablobs?

That makes sense.

heyspacebuck, 2019-06-22:
I have nothing to add other than I'd very much like to see this implemented. A Micronucleus listing within the Arduino Boards Manager would be a HUGE leap forward in user-friendliness. If there's any way I can help make it happen (other than doing all of it myself...believe me, I tried), please let me know.

ArminJo, 2020-05-04:
@Harvie 
Since now it is possible to update the bootloader by the Arduino IDE
[image] 
by using the new [version 1.6.8](https://github.com/ArminJo/DigistumpArduino) of Digistump avr core.
I did not yet get, what you really need?
A 328P support for micronucleus? Using the Arduino/avr package for compiling and then upload the sketch by micronucleus?

@per1234 I think in this case we need introduce a new platform in the [new digispark fork](https://github.com/ArminJo/DigistumpArduino), since we use different compiler?

Jarvis785, 2021-02-15:
Hi @ArminJo , why this issue got closed ?
I want to use atmega328 just like the digispark AtTiny85 board. 
I want to emulate keyboard/mouse as well as upload sketch through micronucleus bootloader to atmega38 without any ftdi ..

Is there alredy a solution for that ?

ArminJo, 2021-02-15:
There is a solution for the original problem 😀

Jarvis785, 2021-02-15:
@ArminJo may i know what's the solution ?

Jarvis785, 2021-02-16:
@ArminJo @heyspacebuck  I want to make a atmega328 board along with micronucleus bootloader.

Basically a board digispark  but want use Atemega328 instead of AtTiny85 so that i can upload sketch without any extra ftdi device and want to emulate Keyboard/Mouse. 

Does anyone have any solution to that ?

ArminJo, 2021-02-16:
Best is to look into the commit history and ask the contributor.

Jarvis785, 2021-02-16:
@ArminJo because you have closed this issue so i thought you must be aware of this

mcer12, 2021-12-23:
@Jarvis785 I want to do the same - design a board with atmega328/atmega168 but with micronucleus so that I don't have to use external FTDI chip. Did you manage to solve this?

Jarvis785, 2021-12-25:
@mcer12   No it didn't worked if you or anyone finds any workaround please let me know :(