micronucleus/micronucleus #143: Use WINUSB instead of libusb-win32

E3V3A, 2019-01-16:
Some time ago there was a nice fork by @mariusgreuel to implement WINUSB support to replace the 3rd party dependency on the obnoxious **libusb-win32** implementation. 

Please consider to implement: 
https://github.com/mariusgreuel/micronucleus/commits/winusb

IDK why a PR was never done...

cpldcpu, 2019-01-17:
Interesting, first time I see it. 

However, it looks like this would increase the size of micronucleus quite a bit due to larger device descriptor strings? Also, there only seems to be an executable for the commandline tool?

E3V3A, 2019-01-20:
>  there only seems to be an executable for the commandline tool?

I guess it's just a recompilation of this, but including the few changes he made? Honestly, I don't know and didn't try it (since I currently have other issues porting another arduino tool using another deprecated usblib ), but seem like a very good idea.

PizzaProgram, 2020-03-09:
[SOLVED]

_(After wasting 5+ hours on trying to fix "libusb-0-1-4.dll missing" error,)_
 This 2018 compiled version **worked** like charm on my Win7 64bit PC :+1: 

Should be included in the MASTER (in a subdir), and in the README file too !!!
 [direct download link of the EXE file...](https://github.com/mariusgreuel/micronucleus/blob/fecb0c4bbe471c10c05de4212aaa533c83271ee5/commandline/builds/Windows/micronucleus.exe)

ArminJo, 2021-02-04:
The bootloader size will increase drastically to support WINUSB. 😢

closed 2021-02-04 by ArminJo