micronucleus/micronucleus #144: USB Example sketches kills bootloader enumeration

E3V3A, 2019-01-16:
I was playing along happily with my newly acquired [Digistump](http://digistump.com/products/1) device, uploading and running sketches directly from Windows (over USB to on-board uUSB) using the `Arduino 1.8.8` IDE. My device doesn't have the PCB-traces based USB, but a proper micro-USB and look like this:

[image]

However, while trying to understand why the Arduino serial console would not work while the device was connected to USB, I decided to try the USB related Digistump *examples*, to see if I would get some kind of a response. So I uploaded the `DigiBlink.ino` example found in the [installed](https://digistump.com/wiki/digispark/tutorials/connecting) library:   

`...\Arduino15\packages\digistump\hardware\avr\1.6.7\libraries\DigisparkUSB\examples\DigiBlink\DigiBlink.ino` example. 

```c
#include <DigiUSB.h>

#define USB_CFG_DEVICE_NAME     'D','i','g','i','B','l','i','n','k'
#define USB_CFG_DEVICE_NAME_LEN 9

byte in = 0;
int Blue = 0;
int Red = 0;
int Green = 0;
int next = 0;

void setup() {
    DigiUSB.begin();
    pinMode(0,OUTPUT);
    pinMode(1,OUTPUT);
    pinMode(2,OUTPUT);
}

void loop() {
  setBlue();
  DigiUSB.refresh();
  setBlue();
  if (DigiUSB.available() > 0) {
    in = 0;
    in = DigiUSB.read();
    if (next == 0){
      if(in == 115){
        next = 1;
        DigiUSB.println("Start");
      }
    }
    else if (next == 1){
      Red = in;
      DigiUSB.print("Red ");
      DigiUSB.println(in,DEC);
      next = 2;
    }
    else if (next == 2){
      Green = in;
      DigiUSB.print("Green ");
      DigiUSB.println(in,DEC);
      next = 3;
    }
    else if (next == 3){
      Blue = in;
      DigiUSB.print("Blue ");
      DigiUSB.println(in,DEC);
      next = 0;
    } 
  }
  analogWrite(0,Red);
  analogWrite(1,Green);
  setBlue();
}

void setBlue(){
    if(Blue == 0){
      digitalWrite(2,LOW);
      return;
    }
    else if(Blue == 255){
      digitalWrite(2,HIGH);
      return;
    }
    // On period  
    for (int x=0;x<Blue;x++){
      digitalWrite(2,HIGH);
    } 
    // Off period
    for(int x=0;x<(255-Blue);x++){
      digitalWrite(2,LOW);
    }
}
```

**This was a stupid mistake!!**  
(There are no multiple USB endpoints for these drivers!)

Because now windows keep thinking this is a `DigiUSB` device and the Arduino no longer recognize the device as the (previous) `Digispark Bootloader`, and refusing to upload any new sketch. Trying to reinstall the drivers, the bootloader is instead found under `"Universal Serial Bus devices"`.

[image]

* Is there any simple way to revert this without a specialized ISP?  
(I.e. Can I use *avrdude* to force upload using the DigiUSB?)

* Where should I report that perhaps the examples (for USB functions) will effectively brick the device from any future new sketch uploads using USB?

E3V3A, 2019-01-16:
This brings me back to (closed) issue #72 which suggested to have a pin to act as a *bootloader enable* during start, to prevent things like this from happening. 

> On boot, the user can hold a button to enter bootloader mode. Otherwise it would act normally.

bratoff, 2019-01-17:
1. Since this is a Chinese mystery board and not a true Digispark, have you verified that the vendor did not leave reset enabled?  If they did, it's just a matter of wiring up an ISP connector and you can reflash a fresh copy of micronucleus, wiping out the problematic code.  
2. If reset is disabled, there are a number of simple HVSP rescue solutions available. Google around and then breadboard one and use it to clear the Disable Reset fuse.  Then you can reflash micronucleus via ISP.  I do a lot of projects with the attiny85 and would not be caught without a HVSP rescue circuit in my bag of tools.

ArminJo, 2019-01-17:
Hi, the arduino delay is way to long, I know.
Therefore I just use this script, to avoid the delay:

```
echo Upgrade Digispark Bootloader with micronucleus upload
%UserProfile%\AppData\Local\Arduino15\packages\digistump\tools\micronucleus\2.0a4\launcher.exe -cdigispark -Uflash:w:upgrade-t85.hex:i
pause
```
You can choose any hex file you want or use the upgrade -** hexfiles to install a fresh micronucleus.
And you can even first start the script and then insert the board.
Good luck.

E3V3A, 2019-01-20:
@ArminJo 
My issue here has nothing to do with the 60s delay, but the <5 sec before the bootloader USB driver changes into DIgiUSB (when the sketch is using it.)

@pfeerick 
Thanks for beautiful explanation, but not quite relevant as above.
Perhaps useful if it was possible to force bootloader use, with some power-on pin settings.

pfeerick, 2019-01-21:
@E3V3A Yeah, I didn't think it was... I just wanted to point out that the delay is only long if there is a) some reason the bootloader isn't working or b) if you plug in the digispark before prompted... hence it will just time out. From what you've said, the USB enumeration delays with Windows is the issue for you, and you're not going to be able to get around that unless you try on a mac/linux box or get HVSP (possibly not necessary if the Chinese have been lazy with attiny85 fuse settings)/ISP working so you can reflash the digispark with a alternate version of micronucleus that does use a jumper to go into the bootloader (which also means no 5 second startup delay).

bratoff, 2019-01-21:
Since you're recompiling the bootloader anyway, why not just increase the boot delay to, say, 10 seconds so Windoze has enough time to enumerate?

E3V3A, 2019-01-22:
@bratoff 
Are you talking to me? 

Where in the sources is that timer set?

pfeerick, 2019-01-22:
@E3V3A If you are just using a custom bootloaderconfig.h, you can simply increase the AUTO_EXIT_MS value from it's currrent default of 6 seconds -  i.e. https://github.com/micronucleus/micronucleus/blob/master/firmware/configuration/t85_default/bootloaderconfig.h#L165

pfeerick, 2019-01-23:
Almost like a double negative... I can see how that would happen. I would think 0 / disable would mean disable the delay. No wait at all. Why? Probably because you would set it to zero if you are using a jumper to enter the reset. Maybe. Possibly. I'd have to read to code to make sure that was the reasoning. :laughing:

stevanuscolonne, 2019-10-10:
Hi all,

Sorry I'm noobs to ATTiny85 programming. I encounter the same issue when mistakenly upload DigiUSB blink into my Digispark ATTiny85 clone and now I cannot simply upload other sketch. Is this issue can be solved with re-flashing the bootloader with ISP? Is there other way than through ISP? Sorry for my bad english since english is not my mother language. Please advise, many thanks in advance.

pfeerick, 2019-10-11:
@stevanuscolonne Your best bet, before requiring either an ISP or perhaps a HV ISP programmer / circuit (depending on the state of the reset pin fuse - it'll either be configured so that the reset pin can still be used for ISP programming, or so that the pin can be used another IO pin, meaning a 'high voltage' programmer / reset is needed) would be to try booting a live version of linux - i.e. run Ubuntu from a flash drive, and trying to re-flash the digispark from there, as it seems the usb enumeration on some windows machines is just too slow to kick in during the five seconds the micronucleus boot-loader is waiting for programming to start.

stevanuscolonne, 2019-10-12:
@pfeerick thank you for your response. I've managed re-flash with ISP. So using arduino ide in linux will be more fast in detecting digiusb as bootloader compare to windows? I familiar with debian since i'm a kali user.

pfeerick, 2019-10-12:
Linux (and probably Mac also) generally seems quicker at detecting/enumerating the device, and passing the message on the micronucleus uploader, so essentially yes, it should be faster than on Windows. 

Glad to hear ISP worked for you... since that worked, if you're not already aware of this, this means that P5 can't be used as an IO pin, since it is still configured for use as the reset pin. 

This is a nice pinout image explaining most, if not all, of the IO quirks of the Digispark ;) https://www.deviantart.com/mortenaaserud/art/Digispark-Attiny85-Pinout-738150591

underwoodblog, 2019-10-13:
Direct link to the image of the digispark:

https://images-wixmp-ed30a86b8c4ca887773594c2.wixmp.com/i/0995f7a6-730b-48d0-8612-c4408d15e84d/dc7h4n3-51e6389c-5f86-4fb2-a7ad-34f75f672003.png