micronucleus/micronucleus #145: How to switch to V2.04 without delay of 6 seconds at startup?

stef-ladefense, 2019-01-19:
Hello,
I'm using an attiny85 digispark module with firmware micronucleus-1.11-entry-ext-reset-upgrade.hex and a push button on P5 to start the download.
I do not see an equivalent version in the 2.04 branch.
How to switch to V2.04 without delay of 6 seconds at startup?
thank

E3V3A, 2019-01-22:
That sound cool. Can you please link to your code?

stef-ladefense, 2019-01-24:
here !!! 
https://github.com/micronucleus/micronucleus/tree/v1.11/upgrade/releases

E3V3A, 2019-01-24:
Thanks! I totally missed the `tags` sub branches. :+1: 
How does your push button work? (P5 to GND or 5V?)

stef-ladefense, 2019-01-24:
GND

it still does not tell me where to find the equivalent with version 2.04

E3V3A, 2019-01-25:
It seem that you are looking for the **sources** for generating that HEX file (above), so that you can update it to 2.04, is that correct? 

PS. I am not affiliated with any of this development, and probably can't help... But I liked the idea of the button.

E3V3A, 2019-01-28:
@Bluebie @cpldcpu 
Any idea where the sources are for all these alternative HEX files?

designer2k2, 2019-02-07:
@E3V3A look at the individual bootloaderconfig.h files: https://github.com/micronucleus/micronucleus/blob/2.04/firmware/configuration/t85_default/bootloaderconfig.h

line 110, ENTRYMODE, set it to ENTRY_JUMPER    to get the Button, or ENTRY_EXT_RESET     for right after reset.

whcrg, 2019-02-13:
ENTRY_JUMPER does not work as expected. (just generated firmware with ENTRY_JUMPER and tried both JUMPER_PIN    PB1 and PB5.

It pulls the specified pin up as expected, but does not care if the pin is grounded or not, just waits and accepts an upload as if the conf was ENTRY_ALWAYS.

whcrg, 2019-02-14:
I was wrong, it does work with PB2 so theres something going on with my physical device that prevented 1 and 5 from working.

pastprimitive, 2019-06-04:
I've been trying to build a successful hex of the ENTRY_JUMPER on PB2. I've changed the following lines from the t85_default bootloader config.h 

From:
line :112 `#define ENTRYMODE ENTRY_ALWAYS`
line :114 `#define JUMPER_PIN    PB0`

To:
line :112 `#define ENTRYMODE ENTRY_JUMPER`
line :114 `#define JUMPER_PIN    PB2`

Everything seems to build fine. I follow the upgrade instructions. Run the ruby script, build the upgrade.hex from the previously built main.hex from the firmware. 

When I upload though pulling PB2 low does nothing. To make sure I was doing everything right. I built a copy of the t85_default with no changes and repeated my procedure. Uploaded it, and it worked as expected. I've even tried using the default PB0 pin. Again no love. 

I really need to ditch the delay for the project, and while I can in a pinch disable the reset on PB5 and use that. I'd rather not. 

@whcrg do you still have the bootloaderconfig.h from your successful PB2 experiment?

whcrg, 2019-06-04:
My build dir with config http://uvkk.kirah.fi/micronucleus/

ArminJo, 2021-01-26:
[Here](https://github.com/ArminJo/micronucleus-firmware) you can find different configurations and especially the fastExit versions, which avoid the 6 seconds timout without requiring to connect the external reset 😀

closed 2021-01-26 by ArminJo