micronucleus/micronucleus #146: Wasting last page of Flash

bobuhito, 2019-01-22:
Seems to me that the last byte in t85_default.hex goes to flash address 0x1FBB,
which is just before the end of the 127th page.  Doesn't this waste one page?
Or, are applications somehow supposed to use this last page?

cpldcpu, 2019-01-22:
ah no. This is, because the bootadress is not updated manually. Due to some new optimizations in AVR-GCC, micronucleus turns out to be a few bytes smaller than at the time of the release. You can safely increase the bootloaderaddress in the Makerfile.inc.

bobuhito, 2019-01-23:
I wouldn't say "a few bytes smaller".  It is actually 68 bytes smaller which is significant.
I know we can change the makerfile manually and compile this manually, but most
people just pull the hex directly and will lose out on those 68 B.

cpldcpu, 2019-01-23:
Well, if it crosses a page border, an additional page has to be used. The previous version was ~10 bytes larger. Need to update it, once I find the time.

ArminJo, 2020-04-30:
You can find the updated version together wit other improvements at https://github.com/ArminJo/micronucleus-firmware

closed 2021-02-04 by ArminJo