micronucleus/micronucleus #148: v 2.4 on t85 is 1786bytes to big for the bootloader? (upgrade from 1.11/2.3)

designer2k2, 2019-02-05:
Hello,
i upgraded from 1.6 to the 1.11 using https://github.com/micronucleus/micronucleus/blob/v1.11/upgrade/releases/micronucleus-1.11-upgrade.hex     
That went smooth, then i took the 2.04 (what tells 2.3 in the debugger): https://github.com/micronucleus/micronucleus/blob/2.04/upgrade/releases/upgrade-t85_default.hex

Again, it worked, and i got 6522Bytes free instead of 6012Bytes.   

So i tried to went to the latest official 2.4 21 Days old from here: https://github.com/micronucleus/micronucleus/blob/2.04/firmware/releases/t85_default.hex

but it fails:
[image]

What can i do to get the 2.4 on my Digistumps?

designer2k2, 2019-02-05:
looking at the hex file in AtmelStudio the start address is already at 1980?
`:1019800017C0197A4005930B09021200010100806B`

where the "old" hex files start at 0000:
`:1000000003C003C003C003C003C003C003C003C0D8`

Ho-Ro, 2019-02-06:
The upgrade-t85_default.hex file differs from the t85_default.hex. The t85*.hex is meant to program via ISP (e.g. STK500) while the upgrade*.hex files can be upgraded via the usb bootloader. You have to create the upgrade* file with the help of the ruby script (or use the python script from my repo https://github.com/Ho-Ro/micronucleus or for those who like to experiment - try to build it with the avr-gcc toolchain as described in https://github.com/Ho-Ro/micronucleus/tree/master/firmware/README.upgrade
HTH, Martin

designer2k2, 2019-02-06:
would it be possible to get all the upgradeX.hex files directly here in this repo?

Im a bit surprised to not get them here, as i thought most users will use the upgrade function instead of the ISP programming (at least it is much easier).

For me i will pass on creating the upgrade.hex files by myself and wait for somebody to build them.

Zitt, 2019-02-07:
I personally don't see the value of having the upgrade files here. Sure; if it's easy fine.
But the author(s) have provided the tools and the means to do this yourself.

For me; I'm programming the boot into devices at Time0 (after the PCB is built); so ISP is really the only thing that is useful for me.

Strictly speaking Digistump (or your uC board manufacturer) should be providing those files for you... not the author(s) of the bootloader code.

Shubhamvis98, 2020-09-05:
I burnt the t85_default.hex in my digispark attiny85
now it is not working anymore even now showing in the device manager
anyone know about how to fix it ...

ArminJo, 2020-09-06:
Please try the files from https://github.com/ArminJo/micronucleus-firmware.

closed 2020-09-06 by ArminJo

Shubhamvis98, 2020-09-06:
when trying t85_default.hex

> Please plug in the device ...
> Press CTRL+C to terminate the program.
> Device is found!
connecting: 20% complete
connecting: 27% complete
connecting: 34% complete
connecting: 40% complete
> Device has firmware version 1.11
> Available space for user applications: 6330 bytes
> Suggested sleep time between sending pages: 8ms
> Whole page count: 99  page size: 64
> Erase function sleep duration: 792ms
parsing: 60% complete
> Program file is 1806 bytes too big for the bootloader!

ArminJo, 2020-09-06:
Please read this issue carefully!

Shubhamvis98, 2020-09-06:
Thanks, bro
it worked

Palmstroemen, 2022-02-08:
It seemed to work. Working on an Win10 System. Installed the driver. Installed the new thing here and then flashed the first DigiSpark to the new bootloader (Recommended). This seemed to work, as then it started to blink rapidly. Something like 3 times each 1/2 second. Which seems to be a sign that no user-program is loaded yet. 

But now as soon as I connect the DigiSpark to the USB it start's blinking almost immediately. And it's not recognized by the IDE. Trying to upload a sketch, or to reburn the bootloader always times out. USB seems to be recognized as there are these Windows-Sounds for adding an USB-device as well as for removing it. 

Any ideas would be welcome.

Shubhamvis98, 2022-02-09:
> It seemed to work. Working on an Win10 System. Installed the driver. Installed the new thing here and then flashed the first DigiSpark to the new bootloader (Recommended). This seemed to work, as then it started to blink rapidly. Something like 3 times each 1/2 second. Which seems to be a sign that no user-program is loaded yet.
> 
> But now as soon as I connect the DigiSpark to the USB it start's blinking almost immediately. And it's not recognized by the IDE. Trying to upload a sketch, or to reburn the bootloader always times out. USB seems to be recognized as there are these Windows-Sounds for adding an USB-device as well as for removing it.
> 
> Connecting another DigiSpark with the old Bootloader ist also not recognized. :-(
> 
> Any ideas would be welcome.

Have you flashed the v1.1 bootloader?
There is an old bootloader which reduces the boot time of attiny85.
So try connecting gnd+p0 while while plugging into your pc while flashing.

ArminJo, 2022-02-09:
@Palmstroemen The reason may be a critical timing, so please try another port, computer or USB hub.

Palmstroemen, 2022-02-09:
I've already tried with gnd->p0. Didn't work. 
I've tried on all USB-Ports direct on my Laptop (Lenovo). Didn't work.
Is there a chance to change the timing somewhere in the configuration?

Palmstroemen, 2022-02-09:
I think I have flashed the 2.5 but not sure. It says "micronucleus.2.5"
No one of my digisparks do respond anymore. Neither those with the old bootloader, nor the one with the new bootloader. 
On the other hand an Arduino Nano with it's AVR-programmer is recognized and is working well. But unfortunately I can't use it as ISP as the DigiSparks do not respond again when attached to the NANO which works as an ISP-Programmer. It's still timing out.
So I think it's maybe something with the driver.

Shubhamvis98, 2022-02-09:
> I think I have flashed the 2.5 but not sure. It says "micronucleus.2.5" No one of my digisparks do respond anymore. Neither those with the old bootloader, nor the one with the new bootloader. On the [...]

Do you have arduino uno or raspberry pi?
Try flashing using spi interface.
I also faced this 

> I think I have flashed the 2.5 but not sure. It says "micronucleus.2.5" No one of my digisparks do respond anymore. Neither those with the old bootloader, nor the one with the new bootloader. On the [...]

Do you have arduino uno or raspberry pi?
Try flashing using spi interface.

Palmstroemen, 2022-02-09:
> Do you have arduino uno or raspberry pi?
> Try flashing using spi interface.
Yes. I have an Uno, and a raspberry pi

Shubhamvis98, 2022-02-09:
> > Do you have arduino uno or raspberry pi?
> > Try flashing using spi interface.
> > Yes. I have an Uno, and a raspberry pi

Then problem solved. Flash isp programmer to arduino uno and flash attiny85 through uno.

Palmstroemen, 2022-02-09:
OK. I'll give it a try.

Palmstroemen, 2022-02-09:
Hm. Ok. It seems that I got the DigiSpark flashed again a new Bootloader via the UNO.
At least the blinking LED scheme has stopped. 
But I still don't get my sketch onto it. Still timing out. As well with the UNO as ISP and the DigiSpark direct connected to the USB-Port of the Computer

Shubhamvis98, 2022-02-09:
> Hm. Ok. It seems that I got the DigiSpark flashed again a new Bootloader via the UNO. At least the blinking LED scheme has stopped. But I still don't get my sketch onto it. Still timing out. As well with the UNO as ISP and the DigiSpark direct connected to the USB-Port of the Computer

I don't get it. Have you tried flashing bootloader using arduino isp?

Palmstroemen, 2022-02-09:
First I have tried to flash a sketch. Didn't work.
Then I tried to flash the bootloader. This seemed to work. Ended with som "successful"-statement. Also the blinking disappeared.
Then tried again to flash a sketch. Didn't work. :-(

How is this process? 
Variant A: Plugoff everything -> Compile, wait for Command to plugin -> Plugin the UNO as ISP + the Digispark?

Variant B: Or Plugoff the DigiSpark from the UNO but leave the UNO with the Capacitor plugged to the computer -> Compile, wait for command to plugin -> Plugin the DigiSpark.

Shubhamvis98, 2022-02-09:
Can I see how you flashing?

Shubhamvis98, 2022-02-09:
> First I have tried to flash a sketch. Didn't work. Then I tried to flash the bootloader. This seemed to work. Ended with som "successful"-statement. Also the blinking disappeared. Then tried again [...]

Variant 2 will work (I guess). And if you successfully uploaded the sketch. Then try flashing the bootloader using command line.