micronucleus/micronucleus #150: make of upgrade fails on os x (probably me doing something wrong)

whcrg, 2019-02-12:
I generated the firmware succesfully with the setting I need (ENTRY_JUMPER with PB5)
but the make of upgrade fails. What am I doing wrong?

Or should I use the 1.* branch upgrade tools, the board is running 1.11 atm?

```
MBM:upgrade jupet$ ruby generate-data.rb ../firmware/main.hex 
const uint16_t bootloader_data[814] PROGMEM = {...};
uint16_t bootloader_address = 6528;
MBM:upgrade jupet$ make clean
rm -f upgrade.hex upgrade.bin *.o upgrade.s upgrade.map upgrade.c.lst
MBM:upgrade jupet$ make
avr-gcc -Wall -Os -fno-move-loop-invariants -fno-tree-scev-cprop -fno-inline-small-functions -I. -Ilibs-device -mmcu=attiny85 -DF_CPU=16500000 -DAPP_ADDRESS=0x80  -c upgrade.c -o upgrade.o -Wa,-ahls=upgrade.c.lst
In file included from upgrade.c:21:
./utils.h:75:5: warning: '_delay_loop_2' is static but used in inline function 'delay' which is not static
     _delay_loop_2((25 * F_CPU / 100000));
     ^~~~~~~~~~~~~
avr-gcc -Wall -Os -fno-move-loop-invariants -fno-tree-scev-cprop -fno-inline-small-functions -I. -Ilibs-device -mmcu=attiny85 -DF_CPU=16500000 -DAPP_ADDRESS=0x80  -o upgrade.bin upgrade.o -Wl,--relax,--gc-sections -Wl,--section-start=.text=80,-Map=upgrade.map
/usr/local/opt/avr-binutils/bin/avr-ld: upgrade.o: in function `beep':
upgrade.c:(.text+0x1ee): undefined reference to `delay'
/usr/local/opt/avr-binutils/bin/avr-ld: upgrade.c:(.text+0x1f6): undefined reference to `delay'
/usr/local/opt/avr-binutils/bin/avr-ld: upgrade.o: in function `main':
upgrade.c:(.text.startup+0xe): undefined reference to `delay'
/usr/local/opt/avr-binutils/bin/avr-ld: upgrade.c:(.text.startup+0x18): undefined reference to `delay'
collect2: error: ld returned 1 exit status
make: *** [upgrade.bin] Error 1
```

whcrg, 2019-02-13:
On ubuntu 18.4 it works so something platform specific?

```
jupet@kek-kone2:~/micronucleus/upgrade$ make
avr-gcc -Wall -Os -fno-move-loop-invariants -fno-tree-scev-cprop -fno-inline-small-functions -I. -Ilibs-device -mmcu=attiny85 -DF_CPU=16500000 -DAPP_ADDRESS=0x80  -c upgrade.c -o upgrade.o -Wa,-ahls=upgrade.c.lst
upgrade.c:172:15: warning: function definition has qualified void return type
 volatile void FakeISR (void) {
               ^
In file included from upgrade.c:21:0:
./utils.h:75:5: warning: ‘_delay_loop_2’ is static but used in inline function ‘delay’ which is not static
     _delay_loop_2((25 * F_CPU / 100000));
     ^
avr-gcc -Wall -Os -fno-move-loop-invariants -fno-tree-scev-cprop -fno-inline-small-functions -I. -Ilibs-device -mmcu=attiny85 -DF_CPU=16500000 -DAPP_ADDRESS=0x80  -o upgrade.bin upgrade.o -Wl,--relax,--gc-sections -Wl,--section-start=.text=80,-Map=upgrade.map
rm -f upgrade.hex upgrade.eep.hex
avr-objcopy -j .text -j .data -O ihex upgrade.bin upgrade-app.hex
cat upgrade-prefix.hex upgrade-app.hex > upgrade.hex
rm upgrade-app.hex
avr-size upgrade.hex
   text	   data	    bss	    dec	    hex	filename
      0	   2356	      0	   2356	    934	upgrade.hex
```

designer2k2, 2019-02-14:
i used an up to date rpi (Raspbian Stretch) and it worked right away to generate the update hex. (2.04)

using this tools: https://github.com/micronucleus/micronucleus/tree/master/upgrade

kugelkopf123, 2019-02-27:
Same Probleme here like TO!

pastprimitive, 2019-06-04:
Exact same readout when making upgrade on OSX 10.14.4.

pastprimitive, 2019-06-04:
After a bit of research from what I could understand, it's an error because of externally calling a function that has a static inline function inside of it. By what I was able to understand a static inline function should not be called externally. I fixed this by changing the following:
Original Line 71 `inline void delay(unsigned int ms) {`
Changed Line 71 `static inline void delay(unsigned int ms) {`

This solved the compilation warning, and so far from my limited testing has not affected anything. Successfully uploaded upgrade.hex to a few digispark clones. 

Hope this helps anyone facing the same problem.

ArminJo, 2021-02-04:
line 71 now is: `void delay(unsigned int ms) {`

closed 2021-02-04 by ArminJo