micronucleus/micronucleus #151: t85_default does not write application

bitflipser, 2019-02-13:
I have put micronucleus 2.04 onto my digispark with 'upgrade-t85_default.hex'. 
Everything seems to be fine: avrdude is doing his job and giving the know protocol on screen when writing a sketch. That works for several trials, but the application will not start.
A flash memory dump with ISP shows, that the sketch isn't burned into flash. The application flash memory is erased (all 0xffff) from 0x000 to 0xcbf (0x197f)* - so the bootloader has erased the memory and kept the comunication protocol with avrdude, but seems to miss the application data (or discards it).
The previous version I uploaded in the middle of october had the same problem.
Is that a known problem?

* no reset vector to the bootloader at 0x000 
  no saved OSCCAL value at -6 (0x197a), 
  no vector to application at -4 (0x197c)

designer2k2, 2019-02-13:
from where do you took the upgrade-t85_default.hex ?

when you already have ISP in place, use: https://github.com/micronucleus/micronucleus/blob/master/firmware/releases/t85_default.hex

bitflipser, 2019-02-14:
Thanks for the fast reply.

Source of 'upgrade-t85_default.hex' is this repository:
<ctrl+k>https://github.com/micronucleus/micronucleus/blob/master/upgrade/releases/upgrade-t85_default.hex</ctrl+k>

and it does its job: 
- the bootloader is placed in the right flash area and is running 
- it correctly signs in as USB-device and communicates with avrdude, e.g. the right amount of free flash for application is reported
- the application flash area is erased (and 'upgrade-t85_default' is wiped out)
- the data packets on USB seem to be acknowledged right, because avrdude is running through to 100%
... but the application area of flash remains empty

So it seems not to be a problem of the upgrade-sketch - it seems to be a problem of 't85_default'.

Why do I want to write the bootloader with an upload sketch instead of ISP? It makes testing of bootloader modifications much easier.

designer2k2, 2019-02-14:
ok, that hex is not the 2.04,  but it is working here for me. (on a official and 2 clone digistumps)
for 2.04 there are currently no upgrade hex files in this repo here 👎  

Please see my issue on that, i pasted a working upgrade-t85 2.04 there: https://github.com/micronucleus/micronucleus/issues/148#issuecomment-461101899

The ISP upload was an idea to test if there is any other issue maybe in place

bitflipser, 2019-02-15:
Ah, you are right. It was version 2.03.
Thanks for giving the version 2.04. 
Installed it, but the song remains the same: the bootloader is placed on the right place, is communicating with avrdude when connected, the application flash are is cleared ... but the application is not written to flash :-( 
Could it be a problem with the new USB-protocol of version 2? Hence version 1.11 is running fine ...

deHarro, 2019-02-15:
I had this problem, too, but I thought I had a faulty Digispark (clone).
I resolved this problem by writing the "self program" fuse with an ISP programmer under avrdude control.

My programmer is "mySmartUSB light" and the accompanying software "myAVR_ProgTool.exe" refuses to read or write the extended fuse. Using avrdude as control program for the flash process resolves this issue.

I already informed the vendor of mySmartUSB light and they told me that the tool offers what Atmel provides in XML descriptors within AtmelStudio. They will fix that when new XMLs are provided by Atmel...

I think I had a second encounter with this problem.
My Digisparks come with micronucleus 1.6 and I upgrade them before I use them.
For upgrading to 2.04 I used the mySmartUSB light with its accompanying software as described above and it seems that (at least sometimes) the "self program" fuse gets reset, resulting in the described problem (all seems to be fine but the user code is not written to flash by micronucleus).

As remedy I use avrdude to upgrade to 2.04 and all seems well up til now.

Harry

bitflipser, 2019-02-18:
Thanks for your reply.
I'm using avrdude when in USB-programming mode (and talking to micronucleus on the digispark):
- with V1.11 - everything is fine
- with V2.04 - app is not written to flash
As far as I know, micronucleus is not reading. reporting or changing the fuse. So, it should not be a problem with the "self program" fuse. 
As ISP I use a self-written (assembler)sketch on Arduino UNO R3. The state ofthe fuse is reported but not change while accessing for reads or writes.

deHarro, 2019-02-18:
Hi bitflipser,
you perhaps got me wrong.
The "self programming" bit seems to be reset by the updating process to V2.04, obviously by means of the programmer software (namely "myAVR_ProgTool") - despite the fact that they claim, their tool cannot touch this fuse (_all_ extended fuses of the Tiny85).

When flashing the new V2.04 under avrdude control (using AvrDudess as GUI), this bit can be and is written correctly.
Reading back this bit with AvrDudess after updating the Chip with "myAVR_ProgTool" results in "0xFF", what means, the self programming bit is cleared (0xFE is the correct value).

Perhaps this has nothing to do with your problem...

Harry

bitflipser, 2019-02-19:
Hi Harry,

thanks for clarifying. Did some testings.
The E-fuse of my digispark ist read as 0x06 (even when it's just written with 0xfe) - I don't know why. 
But at least self programming is enabled ... and the problem is unsolved.

I tried the higher flash program time of 8 ms of micronucleus V1.11 (V2.04 reports 5 ms), but that doesn't help either.

Still searching for a solution.

bitflipser

deHarro, 2019-02-19:
Hi bitflipser,
this gives us a hint at least.
Know that the E-Fuse of a Tiny85 has only one bit to enable/disable, so the only valid values for the E-Fuse are 0xFF and 0xFE, respectively.

Do you have another Digispark to check against? If so, what do you read on that other Digistump?

(I just checked three of my Digistumps and again found one which had the E-Fuse as 0xFF. So I'm led to  assume, this self program fuse thing is not so seldom... in conjunction with my mySmartAVR programmer, at least).

I have to determine that I'm very confident with this tool, even if my last writings seem to state the opposite!! This e-Fuse thing should be gone with the next iteration of the programming software, as stated by the vendor.

Harry

bitflipser, 2019-02-21:
Hi Harry,
I have a clone with micro-USB plug and checked it too. It's exactly the same: V1.11 works fine - V2.04 erases the but not writes the application. 

I don't think it's a matter of E-Fuse setting, because
- 0x06 means SPMEN (bit 0) is programmed (=0)
- avrdude is not able to change that setting while programming over USB
- V1.11 works fine
- V2.04 erases the application area of flash memory (that needs the 'spm'-instruction to work properly)

So, there must be another reason.

bitflipser

cpldcpu, 2019-02-21:
This does not seem right. Can you try V2.03 to make sure it is not an issue with the most recent release?

https://github.com/micronucleus/micronucleus/releases/tag/v2.03

deHarro, 2019-02-21:
Hi bitflipser,
one additional question from me:
Does your Digispark keep its flash when you flash V1.11 _after_ you changed V2.04 back to V1.11 on this very Digispark?

@cpldcpu
I have no such issues with V2.04 on several Digisparks (all with USB as part of the board).
I will check whether it's a matter of Digispark with USB as part of the board or with micro USB plug (I have both types of sparks).
Harry

deHarro, 2019-02-21:
I checked with my Digispark with micro USB port.
It behaves identical to the others with USB port as part of the board.

After flashing V2.04 the spm fuse is reset (I checked this via ISP programming), the spark doesn't get the program (micronucleus.exe states "all well" in this case).
I then set the spm fuse (with avrdudess and mySmartUSB light ISP programmer) and afterwards I can flash the user program via USB (micro USB port).
The user program is retained over several cycles of unplugging/plugging the spark to USB and also if I power the chip over other means.
Hope that helps...
Harry

bitflipser, 2019-02-22:
@cpldcpu
What do you mean by "That doesn't seem right."?

Checked it once more with the linked v2.03. There's no difference between v2.03 and v2.04 (for both t85_default and for t85_aggressive): the application area is erased but the application is not written. 

Changing "Byte 3:  Page write timing in ms" from 5 ms in to 8 ms (as in v1.11) or 4 ms (for testing) does not help.

Still: v1.11 works fine.

Maybe it has to do with the changes made in version 2 . 
With the original version on my digisparks (6.0xx bytes of application space) avrdude used to throw a massege like:
"Oops, lost connection. Need to reconnect. No problem, that happens on some computers sometimes."
after it had counted up the percentages for memory erasing. 

bitflipser

bitflipser, 2019-03-05:
No solution for this problem. I stop searching and use my size-optimized V1,11 (1.412 bytes) instead.

Thanks bitflipser

bitflipser, 2019-03-16:
... final version 1.284 bytes

closed 2021-01-26 by ArminJo