micronucleus/micronucleus #158: comandline shows wrong version

RAWeigel, 2019-07-26:
Sorry, but my english is bad. I use google translator.
In Release 2.04, using the command to view the recorded version, I get the information:
```
> Please plug in the device ...
> Press CTRL+C to terminate the program.
> Device is found!
connecting: 16% complete
connecting: 22% complete
connecting: 28% complete
connecting: 33% complete
> Device has firmware version 2.3
> Device signature: 0x1e930b
> Available space for user applications: 6522 bytes
> Suggested sleep time between sending pages: 7ms
> Whole page count: 102  page size: 64
> Erase function sleep duration: 714ms
> Error opening ../main.hex: No such file or directory
> Error loading or parsing hex file.
```

I use the line:
`... \ micronucleus-master \ commandline \ micronucleus --run --type intel-hex ../main.hex`

Is it my mistake or the new version?

leonty, 2019-09-03:
I mean how to get 2.4 firmware? Which url for the binary?

bryanjhv, 2019-09-04:
Well... First at all, I don't know about you but maintainers made clear that this is NOT a beginner repository, that's the reason why nobody answered to the basic issue in this question.

Anyway, I'm not a maintainer so I'll try to help.

### Firmware upgrade instructions

There are two methods I know for doing this (I assume some knowledge about firmware). Both methods assume that you're using `attiny85` which is the one the DigiSpark uses.

#### Direct upgrade (easier)

Assumes you have `avr-gcc`, `micronucleus` and `ruby` in `$PATH`.

```bash
# Method 1: using precompiled one (not sure it's updated)
wget https://raw.githubusercontent.com/micronucleus/micronucleus/2.04/upgrade/releases/upgrade-t85_default.hex
micronucleus --run upgrade-t85_default.hex
```

```bash
# Method 2: compiling the upgrade yourself
# Source: https://github.com/micronucleus/micronucleus/blob/2.04/upgrade/readme.txt
git clone https://github.com/micronucleus/micronucleus.git
cd micronucleus
git checkout 2.04
cd upgrade
ruby generate-data.rb ../firmware/releases/t85_default.hex
make clean
make
micronucleus --run upgrade.hex
```

#### Firmware flash (harder, and I think safer)

You need a Raspberry Pi or the like for this.
If you have one, try following the tutorial at https://www.instructables.com/id/Refresh-Your-DigiSpark-clone-With-a-Smaller-Bootlo/ but replace the download URL by https://raw.githubusercontent.com/micronucleus/micronucleus/2.04/firmware/releases/t85_default.hex.

leonty, 2019-09-04:
@bryanjhv The issue (both mine and the topic starter's, see the very first description) is that the "micronucleus" utility call designated in the docs (and by you) ALWAYS shows the line "Device has firmware version 2.3". Even after upgrade the firmware from the 2.04 tag.

tomaskovacik, 2019-09-04:
2.3 is the latest bootloader version, so it's at top version

BUT only version 2.0a5 of micronucleus software will upload to this version of the bootloader 

t.

http://kovo-blog.blogspot.com/2019/01/how-to-upgrade-bootloader-on-digistump.html

bryanjhv, 2019-09-04:
2.4 is the latest one. Kindly check.

bryanjhv, 2019-09-04:
@leonty then compile the new binary.

```bash
git clone https://github.com/micronucleus/micronucleus.git
cd micronucleus
git checkout 2.04
cd commandline
make
```

Then use `./micronucleus`. If you're on Windows, maybe download https://github.com/micronucleus/micronucleus/raw/master/commandline/micronucleus.exe and run with that.

leonty, 2019-09-04:
@bryanjhv I'm going to try. Before I used only precompiled binary like you mentioned in the method#1.

bryanjhv, 2019-09-04:
I get it. Usually for now there are no updated binaries for the command line, for example the one that comes from DigistumpArduino is only for 1.x firmware. In Arch Linux I had to install `micronucleus-git` from AUR and symlink the binary to make it work with the new firmware version.

This is the output I'm getting with the CLI from AUR:
(observe the version correctly shows it's 2.4)

```text
# micronucleus --no-ansi --erase-only
> Please plug in the device ... 
> Press CTRL+C to terminate the program.
> Device is found!
connecting: 25% complete
connecting: 34% complete
connecting: 43% complete
connecting: 50% complete
> Device has firmware version 2.4
> Device signature: 0x1e930b 
> Available space for user applications: 6522 bytes
> Suggested sleep time between sending pages: 7ms
> Whole page count: 102  page size: 64
> Erase function sleep duration: 714ms
> Erasing the memory ...
erasing: 75% complete
erasing: 80% complete
erasing: 85% complete
erasing: 90% complete
erasing: 95% complete
erasing: 100% complete
>> Micronucleus done. Thank you!
```

leonty, 2019-09-04:
The method#1:
[image]

Now going to try method#2.

tomaskovacik, 2019-09-04:
Give it some time to flash it self ...

bryanjhv, 2019-09-04:
Ok, I think the first thing you need to figure out is where is the problem: in the micronucleus binary or your device firmware version.
For now I'm gonna assume that 2nd method works, but if not:

```c
#include <usb.h>
#include <stdio.h>

#define VENDOR_ID  0x16D0
#define PRODUCT_ID 0x0753

int find_device() {
  usb_init();
  usb_find_busses();
  usb_find_devices();
  struct usb_bus *bus;
  struct usb_bus *busses;
  busses = usb_get_busses();
  for (bus = busses; bus; bus = bus->next) {
    struct usb_device *dev;
    for (dev = bus->devices; dev; dev = dev->next) {
      if (dev->descriptor.idVendor == VENDOR_ID && dev->descriptor.idProduct == PRODUCT_ID) {
        char major, minor;
        major = (dev->descriptor.bcdDevice >> 8) & 0xFF;
        minor = dev->descriptor.bcdDevice & 0xFF;
        printf("Device version: %d.%d\n", major, minor);
        return 1;
      }
    }
  }
  return 0;
}

int main() {
  int found = 0;
  while (!found) {
    usleep(100*1000);
    found = find_device();
  }
}
```

Please compile that program using: `gcc sample.c -lusb -o sample` (you need to have libusb in your laptop) and run `./sample`, it will wait until you connect the DigiSpark and tell you the version the device itself reports. If it says it's 2.3 then you have the answer to your problem.

bryanjhv, 2019-09-04:
In case it's much work to get rigth `avr-gcc` (see #160), I made upgrade releases which should work for 2.4 firmware. Also, @tomaskovacik is right in that you need to wait some time while it flashes itself (and don't disconnect from the USB).

[upgrade.zip](https://github.com/micronucleus/micronucleus/files/3576189/upgrade.zip)

leonty, 2019-09-04:
I gave it like 10 minutes after flashing.
[image]

bryanjhv, 2019-09-04:
Try with one from the ZIP I uploaded. Usually it should only take a few seconds to flash itself.

leonty, 2019-09-04:
Now it has changed!
[image]

So the issue is that the binary here raw.githubusercontent.com/micronucleus/micronucleus/2.04/upgrade/releases/upgrade-t85_default.hex is outdated?

bryanjhv, 2019-09-04:
You know it's outdated when it was last changed 2 years ago and firmware releases only 8 months ago...

tomaskovacik, 2019-09-04:
So i was right ;) this should be reported :)

leonty, 2019-09-04:
Thank you all!

closed 2021-01-26 by ArminJo