micronucleus/micronucleus #164: [Question] Why is the --run required while updating the firmware ?

Hritik14, 2019-12-01:
I have been trying to flash digispark's firmware and noticed the `--run` option is quite mandatory while applying the firmware to remove 5s delaly.
Why do we need to pass the `--run` argument to micronucleus while uploading the firmware ?
Don't we just want to flash the firmware and not run it at that time ?
Why does it all go crazy (http://digistump.com/board/index.php/topic,320.msg8546.html#msg8546) when we don't provide the --run argument ?
Also, while trying to flash another firmware after this, would I need to connect P0 to GND ?

PizzaProgram, 2020-03-09:
The answer would interest me too...

Ho-Ro, 2020-03-09:
You cannot overwrite the bootloader directly by using the `micronucleus` program because the bootloader is active during this communication - **but** you can upload and *run* a program that by itself overwrites the (now inactive) bootloader.

closed 2020-03-18 by Hritik14