micronucleus/micronucleus #165: Delay at startup

Boia11, 2019-12-06:
Hi,
I wrote a simple code for manage the Led strip brightness with PWM pin of Digisprark ATtiny85 with button attach to interrupt.
On startup it make about 7sec to init code.
I have read that there is a 5 second delay at startup, it's possible remove this delay?

ArminJo, 2019-12-06:
Hi,
It is possible. Do you want to remove the delay at power-up or after reset?
One condition at least is needed to enter programming mode.
Here are the two upgrade hex files

- [upgrade-t85_entry_on_power_on.hex](https://github.com/ArminJo/micronucleus-firmware/blob/master/firmware/upgrades/upgrade-t85_entry_on_power_on.hex). Do not forget in this case to clear the reset flags at startup with `MCUSR = 0; `

- [upgrade-t85_entry_on_reset_no_pullup.hex](https://github.com/ArminJo/micronucleus-firmware/blob/master/firmware/upgrades/upgrade-t85_entry_on_reset_no_pullup.hex)

Boia11, 2019-12-08:
>Do you want to remove the delay at power-up or after reset?
At power-up, I don't use reset feature. I also use "INPUT_PULLUP" inside the code and at power-on attiny don't start.
I need to flash ATtiny85 directly using Arduino Uno as ISP?

I try to flash code using `attiny library by Mellis` with Arduino Uno as ISP and ATtiny work perfectly (No delay and no internal pullup problem)

ArminJo, 2019-12-08:
> I try to flash code using attiny library by Mellis with Arduino Uno as ISP and ATtiny work perfectly (No delay and no internal pullup problem)

Of course it works perfectly without any delay, since it works without bootloader :-)

Boia11, 2019-12-09:
> 
> 
> > I try to flash code using attiny library by Mellis with Arduino Uno as ISP and ATtiny work perfectly (No delay and no internal pullup problem)
> 
> Of course it works perfectly without any delay, since it works without bootloader :-)

Ah okay. Thanks for informations.
But if I use the usb way....whitch bootloader I need to flash for disable delay and use internal pullup?

ArminJo, 2019-12-13:
If you take the `t85_entry_on_reset_no_pullup.hex` one, then you skip the delay at power-up, and enter reprogramming mode (5 seconds delay) by doing a reset.

Boia11, 2019-12-13:
Thanks. I need to install t85_default.hex first, it's correct?
Difference between t85_default and t85_aggressive? Because the aggressive is "critical"?
Can I apply the "upgrade" on both versions?

ArminJo, 2019-12-13:
No other target needs to be installed first. 
I never checked agressive, do you really need the additional flash space on the ATiny???
Upgrades are possible if any working micronucleus is already running on the ATtiny.

Boia11, 2019-12-13:
ok ok. Thanks

closed 2019-12-22 by Boia11