micronucleus/micronucleus #170: Can not install / compile / make under Raspberry Pi Light

PizzaProgram, 2020-03-09:
I've tried to run this program from [OpenHABian image](https://www.openhab.org/docs/installation/openhabian.html).
- I've found no repository for it :(
- Tried to make, but this error appeared:
```
[14:01:58] openhabian@openhab:~/USBi2c/commandline$ sudo make install
[sudo] password for openhabian: ******
make: libusb-config: Command not found
Building library: micronucleus_lib...
gcc  -Ilibrary  -O -g -D LINUX -c library/micronucleus_lib.c
In file included from library/micronucleus_lib.c:30:
library/micronucleus_lib.h:35:10: fatal error: usb.h: No such file or directory
 #include <usb.h>
          ^~~~~~~
compilation terminated.
make: *** [Makefile:56: micronucleus_lib] Error 1
```

scruss, 2020-03-10:
did you install *libusb-dev* as was suggested [here](https://www.raspberrypi.org/forums/viewtopic.php?p=1623998#p1623998)?

PizzaProgram, 2020-03-11:
<OFF>
hmmm... strange!
That message wasn't there last time I've checked. I swear. I've answered the post above that. Probably because I forgot to refresh the page manually in my Browser.
</OFF>

THANKS for notifying me about that, I'll try today or tomorrow. :+1:

Ho-Ro, 2020-03-11:
Just a hint - it helps to [google error massages](http://letmegooglethat.com/?q=fatal+error%3A+usb.h%3A+No+such+file+or+directory) like yours, the first two hits provide the solution.

ArminJo, 2021-02-04:
New binaries are available [here](https://github.com/micronucleus/micronucleus/tree/master/commandline/builds)

closed 2021-02-04 by ArminJo