micronucleus/micronucleus #171: Cannot Detect Device (ATTiny84 on Roshamglo)

twisty-doerofthings, 2020-03-09:
Using Commandline tool version: 2.04 on Lubuntu (```Linux ****** 5.0.0-38-generic #41-Ubuntu SMP Tue Dec 3 00:27:35 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux```).

I can see the device:
```
#:$ lsusb
....
Bus 001 Device 056: ID 1b4f:0014  
....
```
I have created the following file, ```/etc/udev/rules.d/49-micronucleus.rules```, with the content:
```
#UDEV Rules for Micronucleus boards including the Digispark.
# This file must be placed at:
#
# /etc/udev/rules.d/49-micronucleus.rules    (preferred location)
#   or
# /lib/udev/rules.d/49-micronucleus.rules    (req'd on some broken systems)
#
# After this file is copied, physically unplug and reconnect the board.
#
SUBSYSTEMS=="usb", ATTRS{idVendor}=="1b4f", ATTRS{idProduct}=="0014", MODE:="0666"
KERNEL=="ttyACM*", ATTRS{idVendor}=="1b4f", ATTRS{idProduct}=="0014", MODE:="0666", ENV{ID_MM_DEVICE_IGNORE}="1"
#
# If you share your linux system with other users, or just don't like the
# idea of write permission for everybody, you can replace MODE:="0666" with
# OWNER:="yourusername" to create the device owned by you, or with
# GROUP:="somegroupname" and mange access using standard unix groups.
```
Ran command to reload rules: ```udevadm control --reload-rules``` No issues so far. I then run ```micronucleus -``` and see:
```
> Please plug in the device ... 
> Press CTRL+C to terminate the program.

```
I unplug the device and plug it back in with bootload mode (Down on joysitck), see the device light flash indicating it is bootload but the status does not change in micronucleus. It never switches to Connecting.

I can also review ```dmesg``` and see:
```
796811.431980] usb 1-1.4: new low-speed USB device number 62 using ehci-pci
[796811.542571] usb 1-1.4: New USB device found, idVendor=1b4f, idProduct=0014, bcdDevice= 2.03
[796811.542575] usb 1-1.4: New USB device strings: Mfr=0, Product=0, SerialNumber=0
```
I have been unable to find other troubleshooting steps. Arduino and avrdude also been unsuccessful at connecting. More info found here: https://learn.sparkfun.com/tutorials/hack-your-roshamglo-badge

ArminJo, 2021-02-04:
Please test again with the new bootloaders and the new micronucleus binaries

closed 2021-02-04 by ArminJo

twisty-doerofthings, 2021-02-23:
@ArminJo I downloaded and installed the micronucleus-master-LATEST.zip and extracted and ran

`make CONFIG=t84_default`

I then backed up the previous version that was under `/usr/local/bin/micronucleus` and moved the newly created executable to this location. I then tested with the following command:

`micronucleus --help`

I get the following response:

`micronucleus-cli version: 2.5 from Feb 23 2021

usage: micronucleus [--help] [--run] [--dump-progress] [--fast-mode] [--type intel-hex|raw] [--timeout integer] [--no-ansi] (--erase-only | filename)
...
                 filename: Path to intel hex or raw data file to upload,
                           or "-" to read from stdin`

I then tested with a basic command:

`micronucleus -`

And the response:

`Unrecognized option: -`

Will keep testing and looking for new updates. So far not fixed.