micronucleus/micronucleus #172: Make pin behaviour on startup configurable

vogt31337, 2020-04-18:
Hi,

I like the micronucleus bootloader, but I have one serious issue. On startup, the pins are pulled high (during the 5sec booting time). This is problematic, because I have an overdriven LED attached to one pin, and this will fry it.

So is there a way to configure the behaviour of the pins during bootup?

BR

tomaskovacik, 2020-04-19:
resistor in series with led?

ArminJo, 2020-04-19:
Can you please specify 

> the pins

USB pins must be pulled High and Low on startup, or you must choose another [entry condition](https://github.com/ArminJo/micronucleus-firmware#new-start_without_pullup-and-entry_power_on-configurations), not to run into bootloader at power up or reset.

vogt31337, 2020-04-19:
> resistor in series with led
No defies the purpose of overdrive. Idea is to power the LED with too much voltage but for a very brief moment, hence overdrive (perfectly fine for most LEDs).

> the pins

at the moment pin is p2, p1 or p0 would also be possible.
I know p3 and p4 are used by usb and p5 is only accesible via burning some fuses.

I just updated to the newest version of micronucleus and it solved the main issue, as the led is driven but with a very small pulse width. Then again it would be nice to configure the behaviour.

ArminJo, 2020-04-20:
I loaded the [2.5 bootloader](https://github.com/ArminJo/micronucleus-firmware/blob/master/firmware/upgrades/upgrade-t85_default.hex) with [this script](https://github.com/ArminJo/micronucleus-firmware/blob/master/utils/Burn_upgrade-t85_default.cmd), loaded a blink program 
[image]
for P1 and checked the outputs P0 and P2 carefully with my scope.
No driving to high happend, neither after power up, nor after a reset.

vogt31337, 2020-09-06:
I'll try and see if it happens again.

closed 2021-01-31 by cpldcpu