micronucleus/micronucleus #173: Run bootloader only when USB is connected?

gnbl, 2020-04-24:
Is it possible to enter the bootloader on power-on reset only when USB is connected?
I am using the pins `PB3 `and `PB4 `on ATtiny85 only for the USB bootloader (digispark) when updating the firmware (occasionally).
I'd like to eliminate or drastically reduce (<<1 s) the default bootloader time of 6 seconds defined in `AUTO_EXIT_MS`.
I have found the options in https://github.com/micronucleus/micronucleus/blob/master/firmware/configuration/t85_default/bootloaderconfig.h#L80 and was considering using the `ENTRY_JUMPER `or `ENTRY_EXT_RESET `instead of the default. But a solution that does not require additional hardware or interaction would be preferred.
What is the effect of `AUTO_EXIT_NO_USB_MS `and what are recommended values?

<hr/>

After trying to obtain a Make for Windows and getting it running with avr8-gnu-toolchain-win32_x86 I've given up and now use the AVR toolchain installed with Atmel Studio by using `Tools > Command Prompt` to run `make` which seems to set the library paths etc. correctly (they're using Make 4.1).
The Makefile defaults to `t85_default` so `make CONFIG=t85_default` isn't required.

ArminJo, 2020-04-24:
Hi,
I had exact the same problem. I solved it with https://github.com/ArminJo/micronucleus-firmware#new-start_without_pullup-and-entry_power_on-configurations. All I had to add was a reset button.
Good Luck
Armin

gnbl, 2020-04-25:
@ArminJo Thanks, you must be referring to your macros around https://github.com/ArminJo/micronucleus-firmware/blob/master/firmware/configuration/t85_default/bootloaderconfig.h#L158.

I've had a look at these as well but don't know USB; it would be great if the USB line levels could be used alongside PORF:

> • Bit 0 – PORF: Power-on Reset Flag
> This bit is set if a Power-on Reset occurs. The bit is reset only by writing a logic zero to the flag.
> To make use of the Reset Flags to identify a reset condition, the user should read and then reset the MCUSR as early as possible in the program. If the register is cleared before another reset occurs, the source of the reset can be found by examining the Reset Flags.

But I might just modify the macros to allow a high-level button, e.g.
```
// do not enable pull-up, test high level
// untested!
  #define bootLoaderInit()   {JUMPER_DDR&=~_BV(JUMPER_PIN);_delay_ms(1);}
  #define bootLoaderExit()   {JUMPER_PORT&=~_BV(JUMPER_PIN);}
  #define bootLoaderStartCondition() (JUMPER_INP&_BV(JUMPER_PIN))
// untested!
```
Are the AUTO_EXIT_... timeouts in effect for this as well? It would be nice to prevent accidental "hanging" in the bootloader.

ArminJo, 2020-04-26:
Hi,
I enabled the `AUTO_EXIT_NO_USB_MS` [functionality ](https://github.com/ArminJo/micronucleus-firmware/blob/74fc2e64d629678d114a0bfdea3686c60ab28c96/firmware/configuration/t85_fast_exit_on_no_USB/bootloaderconfig.h#L168) and tested it carefully.
The result is that the delay shrinks from 6 seconds do 500 ms. 
Less than 420 ms are not possible!

And of course if you have a spare pin, you can create your own entry conditions like
```
#elif ENTRYMODE==ENTRY_JUMPER
  // Enable pull up on jumper pin and delay to stabilize input
  #define bootLoaderInit()   {JUMPER_DDR&=~_BV(JUMPER_PIN);JUMPER_PORT|=_BV(JUMPER_PIN);_delay_ms(1);}
  #define bootLoaderExit()   {JUMPER_PORT&=~_BV(JUMPER_PIN);}
  #define bootLoaderStartCondition() (!(JUMPER_INP&_BV(JUMPER_PIN)))
```
Cheers
Armin

gnbl, 2020-04-30:
@ArminJo Thanks, you must be referring to your macros around https://github.com/ArminJo/micronucleus-firmware/blob/master/firmware/configuration/t85_default/bootloaderconfig.h#L158.

I've had a look at these as well but don't know USB; it would be great if the USB line levels could be used alongside PORF:

> • Bit 0 – PORF: Power-on Reset Flag
> This bit is set if a Power-on Reset occurs. The bit is reset only by writing a logic zero to the flag.
> To make use of the Reset Flags to identify a reset [...]

But I might just modify the macros to allow a high-level button, e.g.
```
// do not enable pull-up, test high level
// untested!
  #define bootLoaderInit()   {JUMPER_DDR&=~_BV(JUMPER_PIN);_delay_ms(1);}
  #define bootLoaderExit()   {JUMPER_PORT&=~_BV(JUMPER_PIN);}
  #define bootLoaderStartCondition() (JUMPER_INP&_BV(JUMPER_PIN))
// untested!
```
Are the AUTO_EXIT_... timeouts in effect for this as well? It would be nice to prevent accidental "hanging" in the bootloader.

<hr/>

OK so I'm now using 

    #define ENTRYMODE ENTRY_EXT_RESET
    #define AUTO_EXIT_NO_USB_MS    1
    #define AUTO_EXIT_MS           0

this requires a manual reset after connecting to USB, but I can live with that for now.
It think it has started the application code even after that once, but I am not going to investigate. Maybe the value should be increased (I don't want to exit it when USB is connected).
I think there now may be virtually no bootloader delay, I've had to add crude delays to accommodate my hardware requirements.

ArminJo, 2020-04-30:
Does it compile with `#define AUTO_EXIT_MS           0` ?
Better take:
```
#define ENTRYMODE ENTRY_EXT_RESET
#define AUTO_EXIT_NO_USB_MS       0
#define AUTO_EXIT_MS           6000
```
or this [upgrade](https://github.com/ArminJo/micronucleus-firmware/blob/master/firmware/upgrades/upgrade-t85_entry_on_reset_no_pullup.hex)

gnbl, 2020-05-01:
Yes.

ArminJo, 2020-05-01:
IMHO disable means, that it will never leave the bootloader except on receiving an USB-exit command.

cpldcpu, 2020-05-01:
Exactly.

cpldcpu, 2020-05-01:
As a general comment: Detecting presence of an USB connection is very difficult to implement in a robust way that also does not introduce long delays. I would recommend to try to find another entrymode.

One thing that usually works well is to enter the bootload only after external reset. That way the bootloader is never entered if the device is normally powered on.

gnbl, 2020-05-01:
Thanks for the comments.
It works fine with the settings from [above](https://github.com/micronucleus/micronucleus/issues/173#issuecomment-621656317): connect USB (including power), reset to start bootloader, program, disconnect.

Feel free to close this issue - it's too bad github does not include a platform for general discussion / support of a project.

ArminJo, 2020-05-01:
Just to be sure, 
If you reset and are NOT connected to USB will the program start to run after at least 6 seconds?

gnbl, 2020-05-01:
Will test when I get a chance, which might be a while...

closed 2021-01-26 by ArminJo