micronucleus/micronucleus #175: Unreliable Communication - How about designing for a NACK?

bobuhito, 2020-05-16:
Micronucleus v1.11 was apparently slow and caused many NACK responses:
   https://cpldcpu.wordpress.com/2014/03/02/interrupt-free-v-usb
Has anyone checked whether there are still many NACK responses for v2.04
connected to modern PCs?  I see communication problems on about half of
the PCs I connect to, so my suspicion is that many PCs are still too impatient,
but I don't have equipment to monitor these connection like that article used.

It would also be nice to do the CRC/checksum calculation, so I am wondering
what developers would think about intentionally designing for one NACK with
every frame.  In other words, Micronucleus would always send a NACK as an
immediate response, thereby getting double the time for processing, and
then send an ACK on the repeat request.

So, any thoughts on whether that strategy would be significantly more reliable?

(By the way, I see even more strange behavior within loaded programs using
"DigiKeyboard.h", and actually expect that to be worse because it probably
uses some v1.11-like interrupt-style.  Anyway, I wonder if that too could be
improved by designing for one NACK.)

ArminJo, 2020-05-19:
About CRC there you can read in *usbconfig-prototype.h*:
```
#define USB_CFG_CHECK_CRC       0
/* Define this to 1 if you want that the driver checks integrity of incoming
 * data packets (CRC checks). CRC checks cost quite a bit of code size and are
 * currently only available for 18 MHz crystal clock. You must choose
 * USB_CFG_CLOCK_KHZ = 18000 if you enable this option.
 */
```

bobuhito, 2020-05-20:
I guess I should have mentioned that I use ATTINY chips, so can't
get the clock going that fast and wonder if anybody could see what
percentage of frames get NACK responses from these chips.

cpldcpu, 2020-08-23:
The NACKs are sent out by V-USB. The way Micronucleus >2.0 works, it will not send out any NACKs, so the communication speed is improved quite a bit.

Designing software for interrupt free V-USB quite challenging, so most software needs to stay with the normal approach of V-USB, which is to send out a NACK whenever no response is ready.

closed 2020-08-23 by cpldcpu

bobuhito, 2020-08-25:
Why do you say "it will not send out any NACKs"?
Maybe I'm missing something, but I see branches to
sendNakAndReti in the latest assembly code (plus
I have not seen any mention of NACK strategy
changes for newer versions).  Sorry for the trouble,
maybe you can point me to the key assembly code
changes which stopped sending NACKs?

cpldcpu, 2020-08-25:
It's actually explained in the blog article you linked. 

IF V-USB does not have a response assembled yet, it will send out a NACK:
https://cpldcpu.files.wordpress.com/2014/01/all_int.png

Micronucleous will simple igonore the request and respond once it has time. This works only has long as up to two requests from the host have been ignored, so the timing is very critical. The benefit is that no time is wasted on sending out NACKs.

But maybe you are referring to something else?

bobuhito, 2020-08-25:
"Micronucleous will simple igonore the request and respond once it has time."
Instead of ignoring, I was thinking to process for like 20us and then send one
NACK to buy more time.  Then, using that 20us plus the time for the host
to retransmit the request, plus another 20us, the hope is that Micronucleus is
ready to respond.

The 20us could possibly just be 0us and still work, and that was my thinking
when I first posted here.  But if it's somehow easy to code some extra work
before the NACK (so 20us is just a figure for whatever time is convenient),
that would be nice.

cpldcpu, 2020-08-26:
Ah ok, I understand. The thing is, the host will immediately send a new request after receiving a NACK, so only very few cycles are earned. This is not an issue on a normal USB device, since RX/TX are independent of the message processing, but with V-USB it is basically locking the device up. V-USB is usually busy until the end of the frame, when the host pauses for a short while to avoid collision with the frame timing.

An additional complication is that you need an interrupt driven USB stack for the NACK thing to work at all. For reasons described in the article, this is not desired for micronucleus.