micronucleus/micronucleus #177: Device not responding to setup address

redraw, 2020-05-24:
I've bought some years ago, a supposedly Digispark ATTiny85 clone programmer,

[image]

I know little to nothing about programming microcontrollers, just basic concepts. I've read here that I should install micronucleus first in order to then be able to upload sketches using avrdude/Arduino IDE. However, the USB isn't recognized at all, not even reaching udev rules I think. I would try,

```bash
~/tmp/micronucleus $ sudo micronucleus --run firmware/releases/t85_default.hex
> Please plug in the device ... 
> Press CTRL+C to terminate the program.
```

Nothing happens, this is my `dmesg` output when I connect the board,

```
[365083.067222] usb 1-2: new low-speed USB device number 46 using xhci_hcd
[365083.183201] usb 1-2: device descriptor read/64, error -71
[365083.405284] usb 1-2: device descriptor read/64, error -71
[365083.627205] usb 1-2: new low-speed USB device number 47 using xhci_hcd
[365083.742309] usb 1-2: device descriptor read/64, error -71
[365083.965245] usb 1-2: device descriptor read/64, error -71
[365084.073367] usb usb1-port2: attempt power cycle
[365084.701290] usb 1-2: new low-speed USB device number 48 using xhci_hcd
[365084.701839] usb 1-2: Device not responding to setup address.
[365084.905864] usb 1-2: Device not responding to setup address.
[365085.114198] usb 1-2: device not accepting address 48, error -71
[365085.229179] usb 1-2: new low-speed USB device number 49 using xhci_hcd
[365085.229843] usb 1-2: Device not responding to setup address.
[365085.433839] usb 1-2: Device not responding to setup address.
[365085.641221] usb 1-2: device not accepting address 49, error -71
[365085.641378] usb usb1-port2: unable to enumerate USB device
[365106.187222] usb 1-2: new low-speed USB device number 50 using xhci_hcd
[365106.302263] usb 1-2: device descriptor read/64, error -71
[365106.524242] usb 1-2: device descriptor read/64, error -71
[365106.747208] usb 1-2: new low-speed USB device number 51 using xhci_hcd
[365106.862221] usb 1-2: device descriptor read/64, error -71
[365107.084269] usb 1-2: device descriptor read/64, error -71
[365107.193297] usb usb1-port2: attempt power cycle
[365107.821139] usb 1-2: new low-speed USB device number 52 using xhci_hcd
[365107.821662] usb 1-2: Device not responding to setup address.
[365108.025810] usb 1-2: Device not responding to setup address.
[365108.233215] usb 1-2: device not accepting address 52, error -71
[365108.347212] usb 1-2: new low-speed USB device number 53 using xhci_hcd
[365108.347856] usb 1-2: Device not responding to setup address.
[365108.553647] usb 1-2: Device not responding to setup address.
[365108.761132] usb 1-2: device not accepting address 53, error -71
[365108.761285] usb usb1-port2: unable to enumerate USB device
```

ArminJo, 2020-05-24:
Before using a bootloader (here the micronucleus one) it must be flashed onto the chip, which can be done by an ISP and avrdude. You can use the scripts [here](https://github.com/micronucleus/micronucleus/tree/version_2.5/utils).

redraw, 2020-05-24:
Thanks for the quick response, and right, it's a question not a bug.

Just to understand the basics. The idea is that I need to _flash_ the bootloader into the ATTiny85 (using an Arduino as programmer maybe? I'll also check the scripts you mentioned). Then, my Linux machine would recognize the programmer usb board clone with the chip in it?

ArminJo, 2020-05-24:
You are totally right!!! 👍 
I also use only an Arduino as ISP implemented with a Nano on a breadboard.
Good Luck
Armin

redraw, 2020-05-24:
Thx ! Closing it as it was just a question.

closed 2020-05-24 by redraw

drawkula, 2020-05-24:
I'm using a standalone USBASP and a breadboard to flash my chips and have flashed some DIP8-ATtiny85-ers for exactly these boards that way.

redraw, 2020-05-25:
Armin, I've flashed `t85_default.hex` using avrdude and works well. Then upgraded to `upgrade-t85_entry_on_power_on_no_pullup_fast_exit_on_no_USB.hex` using Arduino IDE (Burn bootloader) and your board definitions. and works with USB, but not standalone. I've tried taking out the attiny chip and put it in a simple LED circuit and isn't working. Any guess?

ArminJo, 2020-05-25:
Very strange.
I just tested it with my Digispark and it works. But of course I can not remove the chip.
Can you pull the PB3 (with a resistor) to a VCC or GND and test again? 
Or maybe VCC is below the brownout level?
Or the reset is activated?

redraw, 2020-05-25:
Ok, I applied 5V to the programmer board (without connecting USB) and works. However, it doesn't work with just the chip in a breadboard. I'm sure the circuit in the breadboard works. If I place the LED between chip's VCC and ground through a resistance it turns on, but if I place the LED between Pin 1 and ground doesn't blink. The sketch is just blinking Pin 1.

Oh, I left the breadboard connected while typing, and randomly started running (!) after 5 minutes, though it seems a bit slower. Do you mean trying with a pull up resistor in Pin 3?

redraw, 2020-05-25:
Found out if I drop in/out a resistor between reset and ground it starts, that triggers reset, but not just after connecting the power supply.

ArminJo, 2020-05-25:
> Do you mean trying with a pull up resistor in Pin 3?

Pull up or pull down,both should be OK

redraw, 2020-05-25:
On setup,
```
pinMode(3, INPUT_PULLUP);
```
Sometimes works, sometimes not. What's the purpose of that pullup? Isn't the idea of the bootloader `_no_pullup.hex` to not require it?

redraw, 2020-05-25:
Armin, sorry, this is not the thread to talk about it, I burnt the `t85_entry_on_power_on_no_pullup_fast_exit_on_no_USB.hex` bootloader again with Arduino as ISP.
Flashed a blink sketch, took out the chip, and put it in the breadboard. When I disconnect and reconnect the 5V power supply sometimes starts and sometimes not, I feel like something gets messed up on the bootloader. It's like 6 out of 10 works.
Any guess on that? Am I crazy?

ArminJo, 2020-05-26:
The pullup at PB3 signals that it is a low Speed USB device. Thats all. Only used for USB connection.