micronucleus/micronucleus #178: MH-ET Live Tiny88 support

sergiofagundes, 2020-06-11:
Please, anyone here have a .hex for this board ? I can't find it anywhere on the web.

ArminJo, 2020-06-12:
What board do you have, where is USB- and USB+ connected, Where is the internal LED connected, what is F_CPU?

sergiofagundes, 2020-06-12:
> What board do you have, where is USB- and USB+ connected, Where is the internal LED connected, what is F_CPU?

I have this board:
https://pt.aliexpress.com/item/33014065092.html

I have lost the bootloader and can't program it anymore (unless I use the ISP connector).

ArminJo, 2020-06-12:
I can only help you if you tell me the USB + and USB - connections to the chip or port. They are like here https://s3.amazonaws.com/digispark/DigisparkSchematicFinal.pdf but terminate at different chip pins. You can check it with a continuity tester or an Ohm meter.

sergiofagundes, 2020-06-13:
> I can only help you if you tell me the USB + and USB - connections to the chip or port. They are like here https://s3.amazonaws.com/digispark/DigisparkSchematicFinal.pdf but terminate at different chip pins. You can check it with a continuity tester or an Ohm meter.

I have check with a continuity tester to find these pins, following is what I found:
USB+ PIN1 (D1)
USB- PIN2 (D2)
LED PIN0 D0

ArminJo, 2020-06-13:
You can find the hex file [here](https://github.com/ArminJo/micronucleus-firmware/tree/master/firmware/releases).
If you have problems, please open a new issue at the other repository, (where the hex file is).

closed 2020-06-13 by ArminJo