micronucleus/micronucleus #181: Attiny84 keyboard

Soumojit28, 2020-08-19:
I know a bit of unrelated, but I didn't found any other resource elsewhere. I installed micronucleus in my attiny84 and its working. So seems like it can communicate via USB without an external crystal. Now how I can use this as a keyboard, the digispark keyboard library isn't compatible with the attiny84, so any ideas?

ArminJo, 2020-08-20:
Can you try to port the digispark keyboard library for attiny84. 
Use the [ATTinyCore](https://github.com/SpenceKonde/ATTinyCore) it supports micronucleus.

Soumojit28, 2020-08-20:
`In file included from C:\Users\Soumojit Ash\Documents\Arduino\libraries\DigisparkKeyboard-master\src/DigiKeyboard.h:13:0,

                 from C:\Users\Soumojit Ash\Documents\Arduino\libraries\DigisparkKeyboard-master\examples\Keyboard\Keyboard.ino:1:

c:\program files (x86)\arduino\hardware\tools\avr\avr\include\avr\delay.h:36:2: warning: #warning "This file has been moved to <util/delay.h>." [-Wcpp]

 #warning "This file has been moved to <util/delay.h>."

  ^~~~~~~

In file included from C:\Users\Soumojit Ash\Documents\Arduino\libraries\DigisparkKeyboard-master\examples\Keyboard\Keyboard.ino:1:0:

C:\Users\Soumojit Ash\Documents\Arduino\libraries\DigisparkKeyboard-master\src/DigiKeyboard.h:260:12: warning: multi-character character constant [-Wmultichar]

  if(chr == 'ı') {

            ^~~~

C:\Users\Soumojit Ash\Documents\Arduino\libraries\DigisparkKeyboard-master\src/DigiKeyboard.h:95:1: error: narrowing conversion of '161' from 'int' to 'char' inside { } [-Wnarrowing]

 };

 ^

C:\Users\Soumojit Ash\Documents\Arduino\libraries\DigisparkKeyboard-master\src/DigiKeyboard.h:95:1: error: narrowing conversion of '224' from 'int' to 'char' inside { } [-Wnarrowing]

C:\Users\Soumojit Ash\Documents\Arduino\libraries\DigisparkKeyboard-master\src/DigiKeyboard.h:95:1: error: narrowing conversion of '231' from 'int' to 'char' inside { } [-Wnarrowing]

C:\Users\Soumojit Ash\Documents\Arduino\libraries\DigisparkKeyboard-master\src/DigiKeyboard.h:95:1: error: narrowing conversion of '149' from 'int' to 'char' inside { } [-Wnarrowing]

C:\Users\Soumojit Ash\Documents\Arduino\libraries\DigisparkKeyboard-master\src/DigiKeyboard.h:95:1: error: narrowing conversion of '129' from 'int' to 'char' inside { } [-Wnarrowing]

C:\Users\Soumojit Ash\Documents\Arduino\libraries\DigisparkKeyboard-master\src/DigiKeyboard.h:95:1: error: narrowing conversion of '149' from 'int' to 'char' inside { } [-Wnarrowing]

C:\Users\Soumojit Ash\Documents\Arduino\libraries\DigisparkKeyboard-master\src/DigiKeyboard.h:95:1: error: narrowing conversion of '129' from 'int' to 'char' inside { } [-Wnarrowing]

C:\Users\Soumojit Ash\Documents\Arduino\libraries\DigisparkKeyboard-master\src/DigiKeyboard.h:95:1: error: narrowing conversion of '192' from 'int' to 'char' inside { } [-Wnarrowing]`

any ideas how to fix it? after that I can change the usb configs to match my pin configs

ArminJo, 2020-08-20:
Use the keyboard code from https://github.com/ArminJo/DigistumpArduino (and the updated board manager files).

closed 2020-09-07 by ArminJo